carla-simulator/carla #1727: carla and tensorlayer import conflict: syntax error

hangqiu, 2019-06-07:
Python3 can import carla, tensorflow, tensorlayer individually, but cannot import all of them together. Is there a variable naming conflict?

Version:
Carla: 0.9.5
tensorflow: 1.12.0
tensorlayer: 1.11.0
python: 3.5.2

Python 3.5.2 (default, Nov 12 2018, 13:43:14) 
[GCC 5.4.0 20160609] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import carla
>>> import tensorflow as tf
>>> import tensorlayer as tl
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.5/dist-packages/tensorlayer/__init__.py", line 47, in <module>
    from tensorlayer import initializers
  File "/usr/local/lib/python3.5/dist-packages/tensorlayer/initializers.py", line 7, in <module>
    from tensorlayer.layers.core import LayersConfig
  File "/usr/local/lib/python3.5/dist-packages/tensorlayer/layers/__init__.py", line 31, in <module>
    from .recurrent import *
  File "/usr/local/lib/python3.5/dist-packages/tensorlayer/layers/recurrent.py", line 8, in <module>
    from tensorflow.contrib.rnn import stack_bidirectional_dynamic_rnn
  File "/usr/local/lib/python3.5/dist-packages/tensorflow/contrib/__init__.py", line 57, in <module>
    from tensorflow.contrib import image
  File "/usr/local/lib/python3.5/dist-packages/tensorflow/contrib/image/__init__.py", line 70, in <module>
    from tensorflow.contrib.image.python.ops.single_image_random_dot_stereograms import single_image_random_dot_stereograms
  File "/usr/local/lib/python3.5/dist-packages/tensorflow/contrib/image/python/ops/single_image_random_dot_stereograms.py", line 27, in <module>
    "_single_image_random_dot_stereograms.so"))
  File "/usr/local/lib/python3.5/dist-packages/tensorflow/contrib/util/loader.py", line 56, in load_op_library
    ret = load_library.load_op_library(path)
  File "/usr/local/lib/python3.5/dist-packages/tensorflow/python/framework/load_library.py", line 77, in load_op_library
    exec(wrappers, module.__dict__)
  File "<string>", line 28
    def single_image_random_dot_stereograms(depth_values, hidden_surface_removal=True, convergence_dots_size=8, dots_per_inch=72, eye_separation=, mu=, normalize=True, normalize_max=, normalize_min=, border_level=, number_colors=256, output_image_shape=[1024, 768, 1], output_data_window=[1022, 757], name=None):

                                                                                                                                                 ^
SyntaxError: invalid syntax

gy20073, 2019-06-12:
I got a temporary work around by import carla in the end.

pawel-ziecina, 2019-12-04:
opened ticket on tensorflow https://github.com/tensorflow/tensorflow/issues/34828 project

germanros1987, 2020-06-26:
This should be solved now in newest versions.

closed 2020-06-26 by germanros1987