carla-simulator/carla #1733: New blueprints do not get packaged

ykaganov, 2019-06-10:
I added a new actor blueprint class that inherits from CarlaActorBlueprintFactory that only includes a static mesh. The new blueprint asset is located at "Unreal/CarlaUE4/Content/Carla/Blueprints/Hazards" and the static mesh + material files are located at "Unreal/CarlaUE4/Content/Carla/Static/Hazards". Inside the UE4 editor my blueprint  works fine, but when packaging carla (by running "make package") I don't see the new assets in the "Static" folder. 
What do I need to do in order to package it ?

I also would like to add access to this new actor from the python API. Where do I need to modify the code to obtain this functionality? 

Thank you very much in advance !

nsubiron, 2019-06-11:
Hi @ykaganov,

Have you registered your factory in the _Carla Game Mode_?
[image]

[image]

If your factory is registered and it references your static meshes they should be automatically packaged.

ykaganov, 2019-06-11:
First, Thanks for the quick reply!
I'm using Version 0.9.4 and I see "GameMode: Edit TheNewCarlaGameMode". When I click edit I see a BLANK blueprint. I also switched the GameMode to "CarlaGameMode" and it's also blank. What am I missing ? 

Much appreciate your help !!

nsubiron, 2019-06-11:
It used to be called "TheNewCarlaGameMode" during the transition to the new API, we rename it recently. What do you mean by "blank" blueprint? Have you downloaded the assets package correctly?

ykaganov, 2019-06-11:
Yes, I did download everything correctly before. 
I finally found all of the factories now. It was initially blank (as in an empty page, no info at all). I had to select "Asset -> Reference Viewer" in order to see all the factories. 
I plan to put some pics in a separate comment and hopefully it will help others.
I think this is really important info that should be in the documentation. 

Thanks again !

ykaganov, 2019-06-11:
I had to to reset my layout in the editor. Some windows were hidden. Now I see the menu you showed before. How to point the PropFactory to a new static mesh asset I created?

nsubiron, 2019-06-11:
Check out [How to add custom props to the Blueprint Library](https://carla.readthedocs.io/en/latest/define_custom_props/).

ykaganov, 2019-06-12:
Thanks. Yes this will be great when I upgrade to to 0.9.5.
Meanwhile, In 0.9.4., I managed to add my prop called "tire" to the PropFactory and package everything (see screen shot below).
Only problem is that I cannot find it in the Python API !! What am I missing? 
[image]

Thanks

nsubiron, 2019-06-12:
There was a mistake in that version and there were two lists with props, try in both, I don't remember which one is the right one.

ykaganov, 2019-06-12:
Yep, found it! It's "Definitions" at the bottom of the pic I uploaded. 
I'm grateful for your help !

closed 2019-08-18 by stale[bot]

Thomasochsner, 2025-02-15:
I have the same problem. I add it, but after cooking, it is not changed. What do I need to do?