carla-simulator/carla #1734: Capturing Images along with Lidar PointClouds

vishal-prgmr, 2019-06-10:
Greetings,

How to capture image at the same timestamp of a pointcloud using manual_control.py?? 

Thank You

user025, 2019-06-11:
Try synchronous mode.

vishal-prgmr, 2019-06-11:
@user025 , I would like to use manual_control.py because of all the functionalities. 
However, tried creating a new camera sensor to capture the images in manual_control.py. But calling destroy after capturing the image, leads to no capture of the image. 

````
 if self.sensors[self.index][0].startswith('sensor.lidar'):
           #create new camera sensor
            world = self._parent.get_world()
            bp_library = world.get_blueprint_library()
            camera_bp = bp_library.find('sensor.camera.rgb')
            camera_transform = carla.Transform(carla.Location(x=0.5, z=1.8))
            camera_rgb = world.spawn_actor(camera_bp, camera_transform, attach_to=self._parent)
            channels = image.channels
            pointcount = 0
            for i in range(0, channels):
                pointcount += image.get_point_count(i)
            totalpoints = pointcount*3
            fullbuffer = np.frombuffer(image.raw_data, dtype=np.dtype('f4'), count=-1)
            points = np.frombuffer(image.raw_data, dtype=np.dtype('f4'), count=totalpoints)
            points = np.reshape(points, (int(points.shape[0] / 3), 3))
            if self.recording:
                    x.to_csv(Dataset_full_path.format(image.frame_number), index=None)
                    camera_rgb.listen(lambda data: data.save_to_disk('C:/test/plc_labels/%06d.png' % data.frame_number))
                    camera_rgb.destroy()
````

user025, 2019-06-11:
You may refer to  [how to spawn sensors](https://carla.readthedocs.io/en/latest/python_api_tutorial/#sensors) to see how to get it work.

closed 2019-06-19 by vishal-prgmr