carla-simulator/carla #1737: RPC Error when trying to connect the Python API with the UE4 Editor

pleukefeld, 2019-06-11:
Hi everyone, 

I build the newest version of Carla (0.9.5) from source on Ubuntu 18.04.2.
After starting the Carla simulation in the UE4Editor and running the manual_control.py script, I get the following error:

`Traceback (most recent call last):
  File "manual_control.py", line 861, in <module>
    main()
  File "manual_control.py", line 853, in main
    game_loop(args)
  File "manual_control.py", line 774, in game_loop
    world = World(client.get_world(), hud, args.filter, args.rolename)
RuntimeError: rpc::rpc_error during call in function version`

I also checked that the Ports 2000 and 2001 are free and only used by the UE4Editor.

Do you have a suggestion what might be causing this error? 

Best Regards, 

Paul

osama-usuf, 2019-06-13:
Experiencing the same issue on Windows.

Johnnyswong, 2019-06-30:
Some problem with the client.get_world() calling

eduardohenriquearnold, 2019-07-01:
> Some problem with the client.get_world() calling

Experiencing the same issue building on master's 915e35d435203cb43ec7f462a1c

nsubiron, 2019-07-01:
It works for me on Python3.5, are you sure you built specifically for Python3.6?

eduardohenriquearnold, 2019-07-01:
> It works for me on Python3.5, are you sure you built specifically for Python3.6?

I have compiled the API using `make PythonAPI.3` while having a conda env with Python 3.6 activated. It worked for previous commits (cannot find the last commit it worked, but build tag was 0.9.0-1061-g18c60fba).

nsubiron, 2019-07-01:
Does it generate the egg file with 3.6 tag?

eduardohenriquearnold, 2019-07-01:
Yes, it does.
`creating 'dist/carla-0.9.5-py3.6-linux-x86_64.egg'`

nsubiron, 2019-07-01:
That's correct yes...  not sure what can be causing the RPC call to fail. `rpc::rpc_error during call in function version`, that function returns only a string.

eduardohenriquearnold, 2019-07-01:
Could you reproduce the error?

nsubiron, 2019-07-01:
Sorry, I don't have the bandwidth to test in Python3.6, we only support 3.5 right now.

eduardohenriquearnold, 2019-07-04:
The problem to me was that I was importing the wrong version of the API (I had several compiled ones). And it seems that the latest build method is no longer keeping the api on the path `carla/PythonAPI/build/lib.linux-x86_64-3.6` which used to be standard. I am currently using the .egg file available when you create a package distribution and all seems to be working fine.

lijecaru, 2019-07-25:
I have the same problem, but I am using Python2.7 in Ubuntu 16.04, I built Carla0.9.5 over UE4.22. Any suggestions? I also have the precompiled version installed, could that interfere somehow?

kimiya66, 2019-07-26:
> The problem to me was that I was importing the wrong version of the API (I had several compiled ones). And it seems that the latest build method is no longer keeping the api on the path [...]

sorry I also have this problem now with latest version carla0.9.6 and I could not solve it :( so where do that keep api at the moment in new version? because it never connect to.py files even from carla examples and always get error for client.get_word(). I use python 2.7.15 as i always use this version for carla because it also has problem with python 3....

adi-geek2, 2019-08-02:
There was a way to stop getting RuntimeError: rpc::rpc_error by ensuring 

a.Same version of Python on all machines in n/w
b.Same version of CARLA on all machines in n/w
It worked for me, can someone please confirm that would be great ?

My setup
- Windows local (CARLA 0.9.5- server render machine) - Python 3.7.0
- Linux VM remote (CARLA 0.9.5- PythonAPI client manual_control.py) - Python 3.7.3

eduardohenriquearnold, 2019-08-12:
> I have the same problem, but I am using Python2.7 in Ubuntu 16.04, I built Carla0.9.5 over UE4.22. Any suggestions? I also have the precompiled version installed, could that interfere somehow?

They way I solved it was generating a package with `make package` and then adding the .egg file in `carla/Dist/CARLA_Shipping_0.9.xxx/LinuxNoEditor/PythonAPI/carla/dist/carla-0.9.x-pyX.X-linux-x86_64.egg` to the python path (`sys.path.append`).

cjerome0321, 2019-08-13:
> There was a way to stop getting RuntimeError: rpc::rpc_error by ensuring
>
> a.Same version of Python on all machines in n/w
> b.Same version of CARLA on all machines in n/w
> It worked for me, can someone [...]

Hi @adi-geek2 ,

I am trying to build the same set-up as yours on two separated laptops: 
Windows  (CARLA 0.9.5- server render machine) - Python 3.7.0
Linux  (CARLA 0.9.5- PythonAPI client manual_control.py) - Python 3.5
After I ran:
```
Windows:
.\CarlaUE4.exe  -carla-server
Linux:
python3 manual_control.py --host=192.168.1.18
```
The linux client could not connect to the server and giving a runtime error. I've checked the bandwidth between two machines using iperf, which is around 10 Mbits/s. And I also checked connection from linux to 192.168.1.18:2000 using telnet, which is fine. Could you please kindly tell me how you set up the two machines and get it working? Thanks in advance!!

adi-geek2, 2019-08-14:
Hi ,

Did you check if there any firewall blockages ?

Best Regards,
Aditya Deshpande
+49 1782149231

cjerome0321, 2019-08-14:
> Hi , Did you check if there any firewall blockages ? Best Regards, Aditya Deshpande +49 1782149231
> […](#)

Hi @adi-geek2 ,

Thanks for your quick reply. Now I am also getting the 
```RuntimeError: rpc::rpc_error during call in function version```
I will try python3.7 on linux to see if it works for me.
Thanks!

basavaraj1995, 2019-09-04:
[image]

Use Pycharm and select the correct interpreter as show in image.
It should work

adi-geek2, 2019-09-18:
I realized today that dist folder and eventually the .egg files are built only on building CARLA with 'make package '. Sorry if this is repeat info, but it helped me

lassepe, 2019-09-18:
@nsubiron I think you can close this. Seems like everyone is realizing that the issue was on their side, not on CARLA's

closed 2019-11-24 by stale[bot]

navjotkaurdhoat, 2021-07-20:
> There was a way to stop getting RuntimeError: rpc::rpc_error by ensuring
>
> a.Same version of Python on all machines in n/w
> b.Same version of CARLA on all machines in n/w
> It worked for me, can someone [...]

Hello @adi-geek2, I tried your this proposed solution to make sure the Python and CARLA versions, everything seem fine. But, I am still getting this error. Though earlier, CARLA was working fine. Moreover, for my group members it is working fine, it give RPC error for me only. Any help that you can offer at this moment. Thanks!