carla-simulator/carla #1738: Request to make carla objects Client and World pickleable

gunshi, 2019-06-11:
Hi,
I'm currently working on a multiprocessing-based solution to running CARLA paralelly on a compute-cluster and running into errors mostly because some classes that I import from libcarla (like Client and World) are not pickleable (which is a strict requirement for multiprocessing since it uses pickle for serialization in a Process()).
I can look into this more myself(it may not be very difficult to make them pickleable) and maybe contribute those fixes, but just posting first to see if:
1) this is already being looked into/been attempted, or
2) if the people who're actively working on this codebase have a few suggestions to guide the same
If the fixes can be merged into the release branch as well as the docker images released then that'd be especially great because then I wouldn't have to keep working off off an ad-hoc solution.
Thanks!
Gunshi

nsubiron, 2019-06-13:
Hi @gunshi,

I took a quick look a while ago and it's not an easy task. Most of the objects in our API have a reference to a sort of global manager (per client) that connects to simulator and caches data for fast access. Serializing all these wouldn't be trivial, and the API objects without this global manager are useless (the world is literally just a pointer to this object).

Perhaps you can try to pickle just the necessary to initialize a new client on a different process, ip+port, then share actors by id. Perhaps for convenience we can make pickleable objects like transform, location, and other light-weight objects.

gunshi, 2019-06-13:
Got it, I will look into other options then, thanks!

closed 2019-06-13 by gunshi