carla-simulator/carla #1743: How to only get the bounding box of vehicles that are visible in the front camera?

qingmingjie, 2019-06-13:
We notice that we get all bounding boxes of vehicles in front of the camera, even some of them is not visible in the camera.  but actually we only need the visible car's bounding box, how can we get that?

n1try, 2019-07-03:
Did you find a solution?

harderthan, 2019-07-09:
Hello. Maintainers.

I also need the module, which @qingmingjie told, for trainning my custom object detection model.
If there is no module to do like that in this Carla repository,  could I develop and PR it?

I'm considering to develop the module like following. I will remove not necessary bounding boxes of images which we can get from [client_bounding_boxes.py](https://github.com/carla-simulator/carla/blob/master/PythonAPI/examples/client_bounding_boxes.py).

Maybe, using segmentation information on that's images, I can distinguish with visible object or not. 
If there is some problems to solve this, could you advice better options to me?

noahcao, 2019-07-25:
I'm facing the same requirement, is there any solution till now?

closed 2019-09-30 by stale[bot]

Dumbldore, 2020-01-07:
@qingmingjie  @n1try  @harderthan  @noahcao I am also facing this issue, have any of guys maybe found any good solution?

funatsufumiya, 2023-06-26:
I found the solution. Using segmentation mask and 2d bounding box, you can check bbox is visible from camera.

https://github.com/MukhlasAdib/CARLA-2DBBox
https://carla.readthedocs.io/en/0.9.14/ref_sensors/#semantic-segmentation-camera