carla-simulator/carla #1746: Bottom of the bridge in TOWN5 invalidely classified as ROAD_LINE by semantic segmentation

Ajk4, 2019-06-13:
Reproduction

- I used 0.9.5 release.
- I started server with TOWN5
- I used examples/manual_control.py script
- I lunched application I switched view to semseg using tilde

This is the result I got:

[image]

As you can see bottom of the bridge is marked as Road line class.

XGodina, 2019-06-17:
Hi @Ajk4, There is a problem when imported the mesh from RoadRunner. The bottom of the bridge and lanemarking as the same semantic segmentation.  Thank you for telling us.

germanros1987, 2020-06-26:
Is this still the case @XGodina ?

XGodina, 2020-06-29:
@germanros1987 @Ajk4 Hi, with the new versions of RoadRunner I haven't been able to try it yet. And for the next release, it will be fixed.

closed 2024-04-22 by DnDare