carla-simulator/carla #1748: Can I add bouding-box on my own actor or building?[0.9.5]

ChrisHRZ, 2019-06-13:
I see someone add bounding box for vehicle, can I add bounding box for building or my own actor?

user025, 2019-06-14:
Boundingbox is from UE4 models. CARLA provides this attribution for Vehicles. You can see in  [this document](https://carla.readthedocs.io/en/latest/python_api/#carlavehiclecarlaactor)
Buildings are not modeled this way. You have to figure the bounding box yourself out if you want to add one.

ChrisHRZ, 2019-06-15:
> Boundingbox is from UE4 models. CARLA provides this attribution for Vehicles. You can see in [this document](https://carla.readthedocs.io/en/latest/python_api/#carlavehiclecarlaactor)
> Buildings are not modeled this way. You have to figure the bounding box yourself out if you want to add one.

Thank you! Can you give me more details about figure the bounding box? 
Like should i change carla C++ source code or do something in UE4? Thank you again!

wattsavolt, 2019-06-19:
You would have to add a box component to the actor in the Unreal project code and then expose that actor to the python api. Once you have the bounding box information on the client side, you can refer to client_bounding_boxes.py python example for how to draw it in pygame

ChrisHRZ, 2019-06-21:
@wattsavolt ok. Thanks a lot

remitoudic, 2019-06-25:
hey, 
it might  help you , I found in one from the branches , it works for me fine on 0.9.5 
[bounding_box.txt](https://github.com/carla-simulator/carla/files/3324455/bounding_box.txt)

closed 2019-08-31 by stale[bot]