carla-simulator/carla #1751: add semantic label to lidar point cloud

Tom-Huang, 2019-06-14:
# The projection of semantic label on lidar data is not accurate
I do the following to add semantic labels to point cloud data from lidar sensor in carla:
1.I get the intrinsics as stated in #56.
2.Project 3d points into semantic segmentation image of different views(front, left, right, rear). 

effect:
Totally, the road, buildings and surrounding look good, but the vehicles and pedestrains seem to have some errors. 

The general effect:
[image]

The local accuracy is not accurate:
[image]

The corresponding semantic images of front, left, right and rear views:
front:
[image]
left:
[image]
right:
[image]
rear:
[image]

The problem is: the lidar data points cast onto the vehicle does not have the same shape as the vehicle. As a result, the projection has deviation. Does anybody have any idea of how to improve the accuracy?

user025, 2019-06-14:
#1010  might be helpful.

closed 2019-08-20 by stale[bot]

gnaneshkm, 2019-09-11:
How to get camera intrensic parameters in carla 0.9.5 . I tried to find carlasettings.ini , but i did'nt find . can any one please help me

heshameraqi, 2020-02-06:
Could you please share your code for projection? It will be very helpful.

Tom-Huang, 2020-02-06:
> Could you please share your code for projection? It will be very helpful.

You can check my github repo [carla_data_collection](https://github.com/Tom-Huang/carla_data_collection). Hope it will help!

marcgpuig, 2020-07-31:
This feature has been added in CARLA in #3131, with better performance and more accurate bounces in vehicles and pedestrians.