carla-simulator/carla #1756: automatic_control.py not running despite adding carla directory to PYTHONPATH

programmer6288, 2019-06-14:
Hello,

When I try to run the automatic_control.py script, it shows this error:

> pygame 1.9.6
> Hello from the pygame community. https://www.pygame.org/contribute.html
> Traceback (most recent call last):
> File "automatic_control.py", line 88, in
> from agents.navigation.roaming_agent import RoamingAgent
> ImportError: No module named agents.navigation.roaming_agent

This is the same error that is shown in this issue: [Issue #1543](https://github.com/carla-simulator/carla/issues/1543).

However, after adding the carla directory to my PYTHONPATH, I am still getting the same error.

This is the command I used the add the directory into the PYTHONPATH
`export PYTHONPATH=$PYTHONPATH:carla-0.9.5/PythonAPI/carla/`
and it gave no output

qhaas, 2019-06-25:
It would be nice if 'carla/agents' had a setup.py to deploy it like the core PythonAPI.

closed 2019-07-09 by nsubiron

autonomobil, 2019-08-19:
I had a similar issue in a conda enviroment. This was my solution:
 
- locate .egg-files in `Carla/PythonAPI/carla/dist/`
- change/install python version in conda enviroment which fits closest to an .egg-file (for me carla-0.9.6-py3.5-linux-x86_64.egg -> 3.5)
- delete all other .egg-files (just for tidiness)
- extract your .egg-file, e.g. carla-0.9.6-py3.5-linux-x86_64
- create `setup.py` inside the folder (carla-0.9.6-py3.5-linux-x86_64) with the following content:

```
from distutils.core import setup
setup(name='carla',
      version='0.9.6', #doesn't matter I guess
      py_modules=['carla'],
      )
```
- then you can install the carla-python-package via pip: 
`pip install -e ~/Carla/PythonAPI/carla/dist/carla-0.9.6-py3.5-linux-x86_64`

That solved it for me. I don't even need the egg-path-appending-stuff, which is happening at the beginning of every python-script, just `import carla.`

PR #1859 (2019-07-09): Fix import for automatic_control.py
Due to folder structure, the actual version raises an issue: https://github.com/carla-simulator/carla/issues/1756
It can be fixed using another type of import, however the folder structure has to remain the same for it to work.

#### Description

Changed the import for agents folder due to the structure of this repository, since the other apparently didn't work.

Fixes #1756 

#### Where has this been tested?

  * **Platform(s):** Ubuntu 16.04
  * **Python version(s):** 3.5
  * **Unreal Engine version(s):** 4.42
#### Possible Drawbacks

If the folder structu [... cut]