carla-simulator/carla #1759: Attach sensors to a multi-body vehicle?

pergustav, 2019-06-17:
Hi,

I'd like to attach (multiple) sensors to a multi-body vehicle, where the single sensor's position/fov is affected of the body movement where it's attached.
Is this possible to do?

Thanks for input!

wattsavolt, 2019-06-19:
Currently you can only attach one actor to another actor from the python api. A multi-body vehicle would be made up of multiple static mesh components and each would have its own location and orientation in vehicle space but they belong to the same actor. Therefore what you're looking for is possible in Unreal but our API would have to be updated to allow users to add an actor's root component as a child of one of the components of another actor. Currently we only allow for attaching the root component of one actor to the root component of another actor.

closed 2019-08-26 by stale[bot]

vinayakap, 2021-10-14:
> 
> 
> Currently you can only attach one actor to another actor from the python api. A multi-body vehicle would be made up of multiple static mesh components and each would have its own location and orientation in vehicle space but they belong to the same actor. Therefore what you're looking for is possible in Unreal but our API would have to be updated to allow users to add an actor's root component as a child of any of the components of another actor. Currently we only allow for attaching the root component of one actor to the root component of another actor.

Hi can i attach static mesh to a vehicle via python API, if so,how?