carla-simulator/carla #1762: undefined symbol: PyString_Type

zwbai, 2019-06-17:
after the installation, I want to run the tutorial.py to test the environment. But here is the problem:
[image]
I also find that the problem is whenever I import Carla in python, this problem will occur. is anyone knows why?

sabhiram, 2019-06-19:
Isn't this a duplicate of : https://github.com/carla-simulator/carla/issues/1765?

closed 2019-06-20 by zwbai

havefun28, 2019-06-29:
Hi! Have you solved it? If so, how?

sound-design, 2020-01-30:
Yes it is a duplicate of an issue that is still as of January 29th still unsolved.

revikiran1991, 2020-02-11:
Same here. I am also facing this issue fro a long time. specially when I make own carla