carla-simulator/carla #1763: [0.9.5] make PythonAPI issue - Win10 + Python3.7 + UE4.22

osama-usuf, 2019-06-17:
**OS:** Windows 10
**UE4:** 4.22
**CARLA:** 0.9.5

So I have done a fresh build of the 0.9.5 version of CARLA through the steps provided in the official documentation, but when it comes to building the corresponding PythonAPI ` make PythonAPI`, I am getting the following error:

```
Building Python API for Python 3.
compiling:
  - source/libcarla/libcarla.cpp
  - dependencies/include/carla\Buffer.cpp
  - dependencies/include/carla\Exception.cpp
  - dependencies/include/carla\FileSystem.cpp
  - dependencies/include/carla\StringUtil.cpp
  - dependencies/include/carla\client\Actor.cpp
  - dependencies/include/carla\client\ActorAttribute.cpp
  - dependencies/include/carla\client\ActorBlueprint.cpp
  - dependencies/include/carla\client\ActorList.cpp
  - dependencies/include/carla\client\BlueprintLibrary.cpp
  - dependencies/include/carla\client\DebugHelper.cpp
  - dependencies/include/carla\client\GnssSensor.cpp
  - dependencies/include/carla\client\LaneInvasionSensor.cpp
  - dependencies/include/carla\client\Map.cpp
  - dependencies/include/carla\client\ServerSideSensor.cpp
  - dependencies/include/carla\client\TimeoutException.cpp
  - dependencies/include/carla\client\TrafficLight.cpp
  - dependencies/include/carla\client\Vehicle.cpp
  - dependencies/include/carla\client\Walker.cpp
  - dependencies/include/carla\client\Waypoint.cpp
  - dependencies/include/carla\client\World.cpp
  - dependencies/include/carla\client\detail\ActorFactory.cpp
  - dependencies/include/carla\client\detail\ActorState.cpp
  - dependencies/include/carla\client\detail\ActorVariant.cpp
  - dependencies/include/carla\client\detail\Client.cpp
  - dependencies/include/carla\client\detail\Episode.cpp
  - dependencies/include/carla\client\detail\EpisodeProxy.cpp
  - dependencies/include/carla\client\detail\EpisodeState.cpp
  - dependencies/include/carla\client\detail\Simulator.cpp
  - dependencies/include/carla\geom\GeoLocation.cpp
  - dependencies/include/carla\geom\Math.cpp
  - dependencies/include/carla\opendrive\OpenDriveParser.cpp
  - dependencies/include/carla\opendrive\parser\GeometryParser.cpp
  - dependencies/include/carla\opendrive\parser\GeoReferenceParser.cpp
  - dependencies/include/carla\opendrive\parser\JunctionParser.cpp
  - dependencies/include/carla\opendrive\parser\LaneParser.cpp
  - dependencies/include/carla\opendrive\parser\ObjectParser.cpp
  - dependencies/include/carla\opendrive\parser\ProfilesParser.cpp
  - dependencies/include/carla\opendrive\parser\RoadParser.cpp
  - dependencies/include/carla\opendrive\parser\SignalParser.cpp
  - dependencies/include/carla\opendrive\parser\TrafficGroupParser.cpp
  - dependencies/include/carla\opendrive\parser\pugixml\pugixml.cpp
  - dependencies/include/carla\pointcloud\PointCloudIO.cpp
  - dependencies/include/carla\road\Lane.cpp
  - dependencies/include/carla\road\LaneSection.cpp
  - dependencies/include/carla\road\Map.cpp
  - dependencies/include/carla\road\MapBuilder.cpp
  - dependencies/include/carla\road\MapData.cpp
  - dependencies/include/carla\road\Road.cpp
  - dependencies/include/carla\road\element\LaneCrossingCalculator.cpp
  - dependencies/include/carla\road\element\LaneMarking.cpp
  - dependencies/include/carla\road\element\Waypoint.cpp
  - dependencies/include/carla\road\element\cephes\fresnl.cpp
  - dependencies/include/carla\road\element\cephes\polevl.cpp
  - dependencies/include/carla\rpc\WeatherParameters.cpp
  - dependencies/include/carla\sensor\Deserializer.cpp
  - dependencies/include/carla\sensor\s11n\CollisionEventSerializer.cpp
  - dependencies/include/carla\sensor\s11n\EpisodeStateSerializer.cpp
  - dependencies/include/carla\sensor\s11n\ImageSerializer.cpp
  - dependencies/include/carla\sensor\s11n\LidarSerializer.cpp
  - dependencies/include/carla\sensor\s11n\NoopSerializer.cpp
[... 629 lines cut ...]
NoopSerializer.cpp
dependencies/include/carla\sensor\s11n\NoopSerializer.cpp(15): warning C4646: function declared with 'noreturn' has non-void return type
C:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\VC\Tools\MSVC\14.16.27023\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -Idependencies/include -IC:\Users\osamausf\AppData\Local\Programs\Python\Python37\include -IC:\Users\osamausf\AppData\Local\Programs\Python\Python37\include "-IC:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\VC\Tools\MSVC\14.16.27023\ATLMFC\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\VC\Tools\MSVC\14.16.27023\include" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.6.1\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\cppwinrt" "-IC:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\VC\Tools\MSVC\14.16.27023\ATLMFC\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\VC\Tools\MSVC\14.16.27023\include" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.6.1\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\cppwinrt" /EHsc /Tpdependencies/include/carla\sensor\s11n\ObstacleDetectionEventSerializer.cpp /Fobuild\temp.win-amd64-3.7\Release\dependencies/include/carla\sensor\s11n\ObstacleDetectionEventSerializer.obj /experimental:external /external:I dependencies/include/system /DBOOST_ALL_NO_LIB /DBOOST_PYTHON_STATIC_LIB /DBOOST_ERROR_CODE_HEADER_ONLY /D_WIN32_WINNT=0x0501 /DLIBCARLA_WITH_PYTHON_SUPPORT -DLIBCARLA_IMAGE_WITH_PNG_SUPPORT=true
ObstacleDetectionEventSerializer.cpp
C:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\VC\Tools\MSVC\14.16.27023\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -Idependencies/include -IC:\Users\osamausf\AppData\Local\Programs\Python\Python37\include -IC:\Users\osamausf\AppData\Local\Programs\Python\Python37\include "-IC:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\VC\Tools\MSVC\14.16.27023\ATLMFC\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\VC\Tools\MSVC\14.16.27023\include" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.6.1\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\cppwinrt" "-IC:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\VC\Tools\MSVC\14.16.27023\ATLMFC\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\VC\Tools\MSVC\14.16.27023\include" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.6.1\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\cppwinrt" /EHsc /Tpdependencies/include/carla\sensor\s11n\SensorHeaderSerializer.cpp /Fobuild\temp.win-amd64-3.7\Release\dependencies/include/carla\sensor\s11n\SensorHeaderSerializer.obj /experimental:external /external:I dependencies/include/system /DBOOST_ALL_NO_LIB /DBOOST_PYTHON_STATIC_LIB /DBOOST_ERROR_CODE_HEADER_ONLY /D_WIN32_WINNT=0x0501 /DLIBCARLA_WITH_PYTHON_SUPPORT -DLIBCARLA_IMAGE_WITH_PNG_SUPPORT=true
SensorHeaderSerializer.cpp
C:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\VC\Tools\MSVC\14.16.27023\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -Idependencies/include -IC:\Users\osamausf\AppData\Local\Programs\Python\Python37\include -IC:\Users\osamausf\AppData\Local\Programs\Python\Python37\include "-IC:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\VC\Tools\MSVC\14.16.27023\ATLMFC\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\VC\Tools\MSVC\14.16.27023\inc [... cut ...]
libcarla.obj : error LNK2001: unresolved external symbol "struct boost::python::converter::registration const & __cdecl boost::python::converter::registry::lookup_shared_ptr(struct boost::python::type_info)" (?lookup_shared_ptr@registry@converter@python@boost@@YAAEBUregistration@234@Utype_info@34@@Z)
libcarla.obj : error LNK2001: unresolved external symbol "struct boost::python::converter::registration const * __cdecl boost::python::converter::registry::query(struct boost::python::type_info)" (?query@registry@converter@pyth

c66tang, 2019-06-19:
Same issue here. Looking for a solution.

ferhatmelih, 2019-06-21:
I can suggest that you download the 0.9.5 Release and use the **.egg** file from the downloaded package. You can just put **.egg** file into _PythonAPI/carla/dist/_

closed 2019-08-30 by stale[bot]

pythonCod, 2019-11-16:
did you find any solution?

pengpingliang, 2020-04-28:
@osama-usuf 
I met the same problem at carla0.9.8 & windows 10. 
Could you tell me how you resolved it ?