carla-simulator/carla #1765: import carla error: undefined symbol: PyString_Type

zwbai, 2019-06-18:
Hi, after following the tutorial at [https://carla.readthedocs.io/en/latest/how_to_build_on_linux/](url). I want to run the tutorial.py to test the environment. But there is the problem:
`michael@michael:~$ python3
Python 3.5.2 (default, Nov 12 2018, 13:43:14) 
[GCC 5.4.0 20160609] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import carla
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/michael/.local/lib/python3.5/site-packages/carla-0.9.5-py2.7-linux-x86_64.egg/carla/__init__.py", line 8, in <module>
  File "/home/michael/.local/lib/python3.5/site-packages/carla-0.9.5-py2.7-linux-x86_64.egg/carla/libcarla.py", line 7, in <module>
  File "/home/michael/.local/lib/python3.5/site-packages/carla-0.9.5-py2.7-linux-x86_64.egg/carla/libcarla.py", line 6, in __bootstrap__
  File "/usr/lib/python3.5/imp.py", line 342, in load_dynamic
    return _load(spec)
ImportError: /home/michael/.cache/Python-Eggs/carla-0.9.5-py2.7-linux-x86_64.egg-tmp/carla/libcarla.so: undefined symbol: PyString_Type
>>> 
`
Does anyone knows how to solve this problem? thanks.

closed 2019-06-20 by zwbai

fywu85, 2019-06-25:
I have also encountered the issue. Is there ever a solution to this?

zwbai, 2019-06-25:
Hi, @fywu85 
I disabled the anaconda python path and then enabled the system default python then it worked.

fywu85, 2019-06-25:
@zwbai Thank you! After disabling conda python and recompiling everything from scratches, I got the following error message:
```
fangyu@NODE:~/GitHub/carla$ python3
Python 3.7.3 (default, Apr  3 2019, 05:39:12) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import carla
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/fangyu/GitHub/carla/PythonAPI/carla/dist/carla-0.9.5-py3.7-linux-x86_64.egg/carla/__init__.py", line 8, in <module>
  File "/home/fangyu/GitHub/carla/PythonAPI/carla/dist/carla-0.9.5-py3.7-linux-x86_64.egg/carla/libcarla.py", line 7, in <module>
  File "/home/fangyu/GitHub/carla/PythonAPI/carla/dist/carla-0.9.5-py3.7-linux-x86_64.egg/carla/libcarla.py", line 6, in __bootstrap__
  File "/usr/lib/python3.7/imp.py", line 342, in load_dynamic
    return _load(spec)
ImportError: /home/fangyu/.cache/Python-Eggs/carla-0.9.5-py3.7-linux-x86_64.egg-tmp/carla/libcarla.cpython-37m-x86_64-linux-gnu.so: undefined symbol: _PyThreadState_Current
```

zwbai, 2019-06-25:
@fywu85 is this python3.7 the default python of your OS? I think you can try to use the system default python (python 3.5). Plus make sure you have the right .egg file in  the right place

havefun28, 2019-06-28:
Thank you for the post. I encounter the same problem when running Python3.5 for the [conditional imitation learning repo] (https://github.com/felipecode/coiltraine/blob/master/docs/carla_challenge_coil_baseline.md), which requires conda environment `coiltraine`.  The whole error message is 
```
$ python3 view_model.py -f baselines -e resnet34imnet -cp 180000 -cv 0.9
pygame 1.9.4
Hello from the pygame community. https://www.pygame.org/contribute.html
Traceback (most recent call last):
  File "view_model.py", line 89, in <module>
    import model_view.carla09interface as carla09interface
  File "/home/ruihan/coiltraine/model_view/carla09interface.py", line 72, in <module>
    import carla
  File "/home/ruihan/CARLA_0.9.5/PythonAPI/carla/__init__.py", line 8, in <module>
    from .libcarla import *
ImportError: /home/ruihan/CARLA_0.9.5/PythonAPI/carla/libcarla.so: undefined symbol: PyString_Type
```
Could anyone help or have similar issue?

sound-design, 2020-01-30:
I am having this problem too with 0.9.7.

pravinblaze, 2020-02-14:
Here are some of the things I found out in the process of finding a solution for this:

1. If you are using anaconda (which I was), please make sure anaconda's base environment is disabled by default (conda config --set auto_activate_base false) and make sure you are using a terminal that is not running any conda environment while building PythonAPI or package.

2. If you have any boost installation installed at system level either from the deb store or manually downloaded, then please purge all those installations:
sudo apt-get -y --purge remove libboost-all-dev libboost-doc libboost-dev
sudo rm -f /usr/lib/libboost_*

3. But make sure you have all necessary dependencies for boost installed:
sudo apt-get -y install build-essential g++ python-dev autotools-dev libicu-dev libbz2-dev

4. If you had already done make setup in carla, then remove the boost installation folder inside carla:
rm -rf carla/Build/*
and then do make setup again to rebuild boost inside carla.

5. make PythonAPI in a fresh terminal

6. Start fresh terminal, set PYTHONPATH pointing to the newly created .egg file according to the version of python you wanna use.

That should do it. If it still doesn't solve your issue, then let me know :)

SUNGBEOMCHOI, 2020-04-03:
Try 'sudo python file.py'. I think '.cache' in '/home/username/.cache' is secret directory, so it needs administrator rights.

shiba24, 2020-04-11:
@SUNGBEOMCHOI 
No, I think `sudo python` can ba a workaround because by adding sudo you can call system python, not conda environment.

Still I am looking for a good way for conda to co-exist. It is very painful.

ashtonaureleo, 2020-04-16:
issue still persist. 
(base) ashton@ashton-All-Series:~/CARLA_0.9.5/PythonAPI/examples$ conda activate carla
(carla) ashton@ashton-All-Series:~/CARLA_0.9.5/PythonAPI/examples$ sudo python3 spawn_npc.py -n 80
[sudo] password for ashton: 
Traceback (most recent call last):
  File "spawn_npc.py", line 23, in <module>
    import carla
  File "../carla/dist/carla-0.9.5-py3.5-linux-x86_64.egg/carla/__init__.py", line 8, in <module>
    from .libcarla import *
ImportError: libpng16.so.16: cannot open shared object file: No such file or directory