carla-simulator/carla #1766: Bounding boxs for non-agents

ykaganov, 2019-06-18:
Hi,

I would like to have the bounding-box of actors placed in the scene that are not agents (vehicles/walkers). For example, a tree or a box.
I tried modifying PythonAPI/source/libcarla/Actor.cpp by adding the line in bold:
class_<cc::Actor, boost::noncopyable, boost::shared_ptr<cc::Actor>>("Actor", no_init)
    // work-around, force return copy to resolve Actor instead of ActorState.
    .add_property("id", CALL_RETURNING_COPY(cc::Actor, GetId))
    **.add_property("bounding_box", CALL_RETURNING_COPY(cc::Actor, GetBoundingBox))**
    .add_property("semantic_tags", &GetSemanticTags)
    .add_property("is_alive", CALL_RETURNING_COPY(cc::Actor, IsAlive))

But when reading the bounding box in python I get box-location and box-extent that are all zeros.
How can I insure this info is transferred from the server ? 

Thanks !

Yan

analog-cbarber, 2019-06-27:
Yes, it would be really nice to be able to obtain bounding boxes for static actors.

n1try, 2019-07-03:
Any news on this?

closed 2019-09-08 by stale[bot]