carla-simulator/carla #1769: Do not set ego vehicle as autopilot after replay ends

ilikecyf86, 2019-06-18:
Hi CARLA team:
The current replay API `client.replay_file("recording01.log", start, duration, camera)` will set all vehicles to autopilot after duration. However, sometimes I might do not want to set the ego vehicle as autopilot. Is it possible to provide an API with one more argument such as "id", and the vehicle with the provided id will not be autopilot after the duration ends?

bernatx, 2019-06-25:
Hi, yes, we will add an option to ignore the hero vehicle from the replayed session. That way all other actors will be reproduced but the ego vehicle will be free to drive every time in the same conditions. That means also that the hero vehicle will not continue in autopilot at end of the replay session, and that part is what you request. Sounds that good?

ilikecyf86, 2019-06-25:
Yes that's perfect, thanks for the update!

closed 2019-08-31 by stale[bot]