carla-simulator/carla #1770: when will the car get the traffic light data? and some questions about CARLA

zwbai, 2019-06-18:
Thanks for the open source of CARLA. Here I have some questions about the environment.
Firstly, I am wondering when will the ego-vehicle receive the traffic light information, and can I let the car know the information of the next traffic light instantly when it passed the last intersection? 
Another question is what is the code of autopilot in the manual_control.py? I want to learn more about that.
Finally, can I just control the acceleration manually and let the car do other actions automatically? like lane following and stop at an intersection, etc.
Please help out, thank you very much.

wattsavolt, 2019-06-19:
The car knows what traffic light it is at when it enters the traffic light's trigger box that is placed a few meters before the intersection. 

To enable autopilot, call world.player.set_autopilot(True). This sends a call to the server to tell the vehicle being controlled by the player to use auto-pilot. 

You cannot control the acceleration manually when in auto-pilot. Maybe some semi-autopilot mode could be added in the future or you could modify the code and add one if you want.

To see how the auto-pilot functionality is implemented, check the TickAutopilotController function of the AWheeledVehicleAIController class located in the WheeledVehicleAIController.cpp file.

zwbai, 2019-06-24:
@wattsavolt Thank you so much for your kind and clear reply! Actually, I want to modify a semi-autopilot mode which let the car follow the waypoint automatically and the acceleration and whether to lane change is determined by external input. Is that possible if I don't change the internal C++ file and just use PythonAPI to achieve that? 
Thanks a lot！

closed 2019-08-30 by stale[bot]