carla-simulator/carla #1771: Poor performance in synchronous mode

pawel-ziecina, 2019-06-19:
Hi,
I've tested performance of synchronous mode and got strange results - on all maps, on all sensor sets defined in `PythonAPI/utils/performance_benchmark.py` tool **I receive ~21fps**. Even on LIDAR.

Checked with `Stat UNIT` that `Game` entry takes almost all `Frame` time (see below + also https://docs.unrealengine.com/en-US/Engine/Performance/StatCommands/#unit). May it be an issue with synchronization?

[image]

Carla relese: build downloaded from carla project jenkins server from 2019-06-05 (commit 0faffe2)

Carla server started with:

```sh
./CarlaUE4.sh -fps 10 -benchmark -quality-level=Low
```

Diff comparing used benchmark script and original `PythonAPI/utils/performance_benchmark.py`:

```diff
diff --git a/PythonAPI/util/performance_benchmark.py b/PythonAPI/util/performance_benchmark.py
index d15835e8..cb151b9f 100755
--- a/PythonAPI/util/performance_benchmark.py
+++ b/PythonAPI/util/performance_benchmark.py
@@ -197,6 +197,7 @@ def run_benchmark(world, sensor_specs_list, number_locations, number_ticks, acto
 
         ticks = 0
         while ticks < number_ticks:
+            world.tick()
             _ = world.wait_for_tick(1000.0)
             if debug:
                 print("== Samples {} / {}".format(ticks + 1, number_ticks))
@@ -284,6 +285,9 @@ def main(args):
     records = {}
     for town in sorted(client.get_available_maps()):
         world = client.load_world(town)
+        settings = world.get_settings()
+        settings.synchronous_mode = True
+        world.apply_settings(settings)
 
         # spectator pointing to the sky to reduce rendering impact
         spectator = world.get_spectator()

```

Full results: [xserver_low_synch.txt](https://github.com/carla-simulator/carla/files/3305220/xserver_low_synch.txt)

nsubiron, 2019-06-26:
Hi @pawel-ziecina,

Synchronous mode is much slower, that's why we have the distinction. Think that in asynchronous mode the game and render threads are executed in parallel, the renderer is actually executed a frame or two behind the game thread, this way there is always work to do. But in sync mode, the game thread has to wait until the client sends a confirmation to continue, then the render thread has to wait until the game thread generated a new frame, thus a big part of the time they're just waiting idle.

The use case for the sync mode is when your client is slower than the simulator, then you want the simulator to wait for your client.

closed 2019-09-01 by stale[bot]

sapirbm, 2020-01-12:
hey @nsubiron,

a question, if a run carla in sync mode and  **world_settings.no_rendering_mode=True** i should assume that the render thread will not be running, and carla performance will be faster in sync mode. but that's not the case. 
any insights?