carla-simulator/carla #1772: Launching Carla Server keeps Camera at 0.0.0 position - Unable to navigate

chandeep11, 2019-06-19:
HI,

After building the package using **make package**, while launching the server, the camera position is fixed at 0.0.0 position. I'm unable to navigate in the server window. Running manual control.py works fine where I can drive in the map. Even in the Carla UE4 editor, I'm facing the same issue while hitting the play button using the "selected viewport" option. Any solution for this?

wattsavolt, 2019-06-19:
Are you running on Windows or Linux and is this with Carla 0.9.5 release or a build from master?

chandeep11, 2019-06-19:
HI @wattsavolt, I'm currently using Carla 0.9.5 release in Linux which is compiled from source.

wattsavolt, 2019-06-20:
Try the release package directly and see if you get the same result. I couldn't reproduce with that.

chandeep11, 2019-07-01:
Hi @wattsavolt, release package is working fine in our case. 

1.Previously when we setup Carla, we were able to navigate Camera in the world using the "Launch/Play from Selected Viewport" option.

2.Now we are editing the existing map (roads and static assets) in UE4 editor. When we try to spawn any vehicles using the CarlaGamemode in World settings, we are unable to move both the vehicle and the camera as it starts the initial gameplay inside the car with the camera positioned at vehicle's origin. 

3.When we try to set the CarlaGamemode to None and then relaunch the gameplay using "Launch/Play from Selected viewport", the camera is positioned at the world's origin and we are unable to navigate the camera within the map. After clicking "Eject" when we are in simulation, we are able to navigate the camera in the world. 

I'm uploading the issues as video files. Video 1 refers to points 1 and 2. video 2 refers to the point 3.

[issuevideo1.zip](https://github.com/carla-simulator/carla/files/3345302/issuevideo1.zip)
[issuevideo2.zip](https://github.com/carla-simulator/carla/files/3345304/issuevideo2.zip)

aadityasp1, 2019-07-11:
Hi @wattsavolt, any idea how to fix this issue?

aadityasp1, 2019-07-22:
@nsubiron can you help me fix this ?

closed 2019-09-27 by stale[bot]