carla-simulator/carla #1774: [Windows] project fails to launch with editor after build

ykaganov, 2019-06-19:
I follows all the instructions at https://carla.readthedocs.io/en/latest/how_to_build_on_windows/ very very closely.
I am trying to build Carla 0.9.5 (latest) with "make launch" using unreal 4.22.2 and Visual Studios 17.
I have all prerequisites installed (make, git, python 3.6 (x64)).
I am using the x64 Native Tool Command Prompt for Visual Studios 2017.
The build itself is completed successfully, but when opening the project with the unreal editor I receive the message : "The following modules are missing or built with a different engine version: CarlaUE4, Carla \n Would you like to rebuild them now?". After pressing "Yes" and get this message: "CarlaUE4 could not be compiled. Try rebuilding from source manually".
I tried "make rebuild" but it does not help.
What could I be missing ?!!

wattsavolt, 2019-06-19:
Did you clone the Unreal Engine repo and build it? If so, did you switch to the 4.22.2-release branch before building?

ykaganov, 2019-06-19:
I'm on Windows so I installed unreal 4.22.2 from epic's app launcher. Do I have to build the engine from source?

ykaganov, 2019-06-19:
Ok, I generated a visual studio solution from the uproject file and then I found cause of my error.
The xml file here %appdata%\Unreal Engine\UnrealBuildTool\BuildConfiguration.xml was messed up.
The correct format should be:
<?xml version="1.0" encoding="utf-8" ?>
<Configuration xmlns="https://www.unrealengine.com/BuildConfiguration">
    <VCProjectFileGenerator>
        <Version>VisualStudio2017</Version>
    </VCProjectFileGenerator>
    <WindowsPlatform>
        <Compiler>VisualStudio2017</Compiler>
    </WindowsPlatform>
</Configuration>

closed 2019-06-19 by ykaganov