carla-simulator/carla #1777: Wait for tick timeout

Johnnyswong, 2019-06-20:
Hi, @nsubiron 
After the client call the tick function, it seemly stuck in the function wait_for_tick().
I have no idea how to fix this issue.
The following picture shows my code for the synchronous mode.
[image]

ykarmesh, 2019-06-22:
Hey @Johnnyswong, I am also facing issues with wait_for_tick() function (issue #1795). Does your code run at all or does it get struck the first time you try wait_for_tick()? Also, which version of CARLA are you running?

Johnnyswong, 2019-06-23:
Hi @ykarmesh 
I tried the multiprocess framework "ray" to fork 2 simulators. After several steps, one of the simulators hanged at the wait for tick function. My version of CARLA is 0.9.5.

ykarmesh, 2019-06-23:
Although I am not using Ray, the problem is exactly what I am also encountering.

Johnnyswong, 2019-06-24:
> Although I am not using Ray, the problem is exactly what I am also encountering.

Yeah. After some trial and error, I find that there are three reasons might cause this error:
1) The destruction of sensors, collision sensor esp. ;
2) Loading map, esp loading Town04 ;
3) Render timeout;
These reasons can be traced by the log of the simulator.

I tried: 1) remove the collision sensor with my own implementation 2) try to load Town03 3) Disable rendering
Now  it seems ok. It has been running for  3hours and I havn't met this problem.

Johnnyswong, 2019-06-24:
#1795

closed 2019-08-30 by stale[bot]