carla-simulator/carla #1778: Assertion failed when run CppClient application on windows

rajesh8646, 2019-06-20:
Windows 10
Carla version 0.9.5
Unreal version 4.21

I am using the example created for CppClient  using Carla C++ API. The application is build successfully on windows 10.

Also I tried with API "GetWorld" but still same error 

Assertion failed: (_data.size() - _offset) % sizeof(T) == 0u,
\carla\LibCarla\cmake\..\source\carla/sensor/data/Array.h, line 137

Please let me know if I missing any steps.

nsubiron, 2019-06-20:
That probably means that there is a size mismatch between types in server and client, (e.g. one of the types we serialize has different size in bytes in server than in client). For performance, we just reinterpret the data as it comes from the server. We're interested in seeing in which platforms this may fail and try to fix it.

Are you running the server also in Windows? Are both compiled with the same settings?

rajesh8646, 2019-06-21:
Thank you @nsubiron  

Yes, I am running the server ( active map Town03) also on windows 10 64 bits. And build Client using Visual studio on windows.

Serve is build as steps mentioned in below
https://carla.readthedocs.io/en/latest/how_to_build_on_windows/

While building the Carla client in Release mode:

add Preprocesssor:
WIN32
_WINDOWS
_WIN32_WINNT=0x0501
BOOST_ERROR_CODE_HEADER_ONLY
LIBCARLA_IMAGE_WITH_PNG_SUPPORT
CMAKE_INTDIR="Release"
_CRT_SECURE_NO_WARNINGS

added libraries:
carla_client.lib
libboost_date_time-vc141-mt-x64-1_69.lib
libboost_filesystem-vc141-mt-x64-1_69.lib
libboost_numpy37-vc141-mt-x64-1_69.lib
libboost_python37-vc141-mt-x64-1_69.lib
libboost_system-vc141-mt-x64-1_69.lib
libpng.lib
zlib.lib
zlibstatic.lib
rpc.lib
ShLwApi.Lib ( added because I was getting error: error LNK2019: unresolved external symbol __imp_PathMatchSpecA)

Runtime library  - Multi-thread DLL (/MD)

nsubiron, 2019-06-27:
Hi @rajesh8646,

Thanks for the info! could you try adding these asserts in _ActorDynamicState.h_ and see if any of them fail?
```cpp
static_assert(sizeof(float) == 4u, "Invalid size!");
static_assert(sizeof(detail::PackedVehicleControl) == 19u, "Invalid size!");
static_assert(sizeof(detail::PackedWalkerControl) == 17u, "Invalid size!");
static_assert(sizeof(detail::TrafficLightData) == 22u, "Invalid size!");
static_assert(sizeof(detail::VehicleData) == 29u, "Invalid size!");
static_assert(sizeof(ActorDynamicState) == 93u, "Invalid size!");
```

I think the compiler might not be generating the same size for one of these objects in client and server.

manfred-exz, 2019-07-15:
### Problem
Check for `LibCarla/source/carla/sensor/data/Array.h`

``` cpp
template <typename T>
  class Array : public SensorData {
  public:
...
  protected:
    explicit Array(size_t offset, RawData data)
      : SensorData(data),
        _data(std::move(data)) {
      SetOffset(offset);
    }

    explicit Array(RawData data)
      : Array(0u, std::move(data)) {}

    void SetOffset(size_t offset) {
      DEBUG_ASSERT(_data.size() >= _offset);
      DEBUG_ASSERT((_data.size() - _offset) % sizeof(T) == 0u);
      _offset = offset;
      DEBUG_ASSERT(begin() <= end());
    }

    const RawData &GetRawData() const {
      return _data;
    }

  private:

    size_t _offset;

    RawData _data;
  };
```

in `explicit Array(size_t offset, RawData data)`, `_offset` is not initialized, and in `void SetOffset(size_t offset)` it is checked(DEBUG_ASSERT) without initialization.

### Solution
The Solution is to set _offset before check.
```
    explicit Array(size_t offset, RawData data)
      : SensorData(data),
		_offset(offset),
        _data(std::move(data)) {
      SetOffset(offset);
    }

    void SetOffset(size_t offset) {
	  _offset = offset;
      DEBUG_ASSERT(_data.size() >= _offset);
      DEBUG_ASSERT((_data.size() - _offset) % sizeof(T) == 0u);
      DEBUG_ASSERT(begin() <= end());
    }
```

And you have to fix some code in `LibCarla/source/carla/sensor/data/RawEpisodeState.h` and `LibCarla/source/carla/sensor/data/LidarMeasurement.h`.

```
    explicit LidarMeasurement(RawData data)
      : Super(Serializer::GetHeaderOffset(Super::GetRawData()), std::move(data)) {
    }

    explicit RawEpisodeState(RawData data)
      : Super(Serializer::header_offset, std::move(data)) {
    }
```

nsubiron, 2019-07-16:
Totally true! it's checking the offset before initializing :woman_facepalming: 
```cpp
void SetOffset(size_t offset) {
  DEBUG_ASSERT(_data.size() >= _offset);
  DEBUG_ASSERT((_data.size() - _offset) % sizeof(T) == 0u);
  _offset = offset;
  DEBUG_ASSERT(begin() <= end());
}
```

Your solution for the Lidar measurement is broken though, you're not sure in which order the arguments are going to execute thus you could get an uninitialized raw data from the parent class.

manfred-exz, 2019-07-16:
Right, didn't notice that. 

I think the current carla::sensor::data::Array's constructor is quiet hard to use, when you have inheritance and `std::move` at the same time(LidarMeasurement).
```cpp
explicit LidarMeasurement(RawData data)
     : Super(Serializer::GetHeaderOffset(data), std::move(data)) {} // wrong, don't know evaluation order
```

Might be better to have a default constructor and expose _offset/_data to child class.

marcgpuig, 2019-07-18:
The assertion error has been fixed

stavBodik, 2019-12-04:
> Hi @rajesh8646,
> 
> Thanks for the info! could you try adding these asserts in _ActorDynamicState.h_ and see if any of them fail?
> 
> ```c++
> static_assert(sizeof(float) == 4u, "Invalid size!");
> static_assert(sizeof(detail::PackedVehicleControl) == 19u, "Invalid size!");
> static_assert(sizeof(detail::PackedWalkerControl) == 17u, "Invalid size!");
> static_assert(sizeof(detail::TrafficLightData) == 22u, "Invalid size!");
> static_assert(sizeof(detail::VehicleData) == 29u, "Invalid size!");
> static_assert(sizeof(ActorDynamicState) == 93u, "Invalid size!");
> ```
> 
> I think the compiler might not be generating the same size for one of these objects in client and server.

I belive in my case this is exactly what happens, client and server both compiled in windows.
The server is after UE4 compilation and the Client me created new visual studio 2017 project and included all needed files in order to test.

In my case ::
_Data.size() =  8483
offset = 0
sizeof(T) = 93 (ActorDynamicState)

When loading "/Game/Carla/Maps/BaseMap/BaseMap"

Anything I can check/Do in order to fix this ?

stavBodik, 2019-12-05:
Problem solved I found that i should use : Use Multi-Byte Character Set , in the client and the server .

closed 2020-06-26 by germanros1987

inv-midhunbenny, 2022-10-12:
@germanros1987 @nsubiron @stavBodik @Axel1092 @bernatx @marcgpuig 
Running Carla 0.9.13 on Windows 10 build from source as per the documentation using .whl file `carla-0.9.13-cp37-cp37m-win_amd64`(https://carla.readthedocs.io/en/0.9.13/build_windows/#part-two-build-carla)
But after adding the codes for safe distance sensor from the GitHub code given by @nsubiron (https://gist.github.com/nsubiron/011fd1b9767cd441b1d8467dc11e00f9)

I complied the code using the  `make PythonAPI ` and then `make launch`. 

Added the SafeDistanceClass in manual_control.py given in the Carla PythonAPI examples
```
class SafeDistanceSensor(object):
    def __init__(self, parent_actor):
        self.sensor = None
        self._parent = parent_actor
        world = self._parent.get_world()
        bp = world.get_blueprint_library().find('sensor.other.safe_distance')
        self.sensor = world.spawn_actor(bp, carla.Transform(), attach_to=self._parent)
        self.sensor.listen(lambda event: SafeDistanceSensor.callback(world, event))
    @staticmethod
    def callback(world, event):
        for actor_id in event:
            vehicle = world.get_actor(actor_id)
            print('Vehicle too close: %s' % vehicle.type_id)
```
    
 Upon executing the code and driving my car to a near by vehicle the server crashes and assertation error is been logged. 
    `Assertion failed: px != 0, file C:\workspace\carla\Build\boost-1.72.0-install\include\boost/smart_ptr/shared_ptr.hpp, line 734 `

jcontrolresearch, 2023-01-17:
> @germanros1987 @nsubiron @stavBodik @Axel1092 @bernatx @marcgpuig Running Carla 0.9.13 on Windows 10 build from source as per the documentation using .whl file [...]

I had the same issue, there are any fix to this?

xuq9908, 2023-05-17:
> I had the same issue, there are any fix to this?

Hi, I met the same problem. Have you fix this? I pressed play and ran the commands according to windows build document... And then this error was reported