carla-simulator/carla #1779: Adding a custom car

kasperhaak, 2019-06-20:
Hi,

I've tried adding a car custom car to the Carla game. However whenever i start the simulation with the custom vehicle, the car spawns, falls to the ground and then bounces off never to return again. I can't really figure out whats going on, that's why i'm turning here for help.

Some info about the car:
- It is within the 10.000 - 17.000 triangle limit
- The bone names are Base_Vehicle, Front_Left_Wheel, Front_Right_Wheel, Rear_Left_Wheel, Rear_Right_Wheel
- The Base_Vehicle bone is setup at the origin (0,0,0)
- The bone positions of the wheels are set up in the middle of the wheels
- The model and the bones are oriented to the positive X-axis and Z-axis up 

Furthermore I followed the How to add assets guide in order to model the blueprints.
I can't figure out why the car is flying off after it reaches the ground. Any help is much appreciated.

Thanks in advance

closed 2019-08-26 by stale[bot]

prashantluitel, 2019-09-26:
I have the same issue

krdindorkar, 2019-12-23:
@kasperhaak @prashantluitel I am having same issue. Did you resolved yours? Kindly help me

peclatj, 2022-10-26:
Hi, I have the exact same issue after following the guide to add a vehicle.
I tried to change the collision box but it change nothing.

The vehicle spawn and start gaining speed by itself, until it rolls over and fly off in the sky.

peclatj, 2022-11-09:
I found a solution, It has to do with the armature of my mesh in blender. I have no idea why but mine does not work but the armature provided (https://carla.readthedocs.io/en/latest/tuto_A_add_vehicle/) on the 2nd tutorial works, I'm unable to find a single difference in the parameters in blender however.