carla-simulator/carla #1782: Is there any reinforcement learning tutorial for CARLA 0.9.X ?

zwbai, 2019-06-20:
Hi,

Thanks for the remarkable work of CARLA simulator.
I am now wondering how to build a reinforcement learning training environment based on CARLA 0.9.5. But so far most of the RL environment is based on CARLA 0.8.X which is much more different with 0.9.X version. 
Thus, is there any 0.9.X based RL environment tutorial? or how to build that? 
Thank you very much!

Johnnyswong, 2019-06-23:
hi @zwbai 
You can use the env wrapper in openai's baselines to get a quick start(https://github.com/openai/baselines).
OpenAI Gym provide a easy way to implement a new environemnt, you only need to write several functions, eg. env.reset(), env.step(action).

zwbai, 2019-06-24:
Hi @Johnnyswong thank you very much, but I did not find any tutorial of how to build an RL environment at [https://github.com/openai/baselines](url). Can you give some more specific information?
Thanks a lot!

osama-usuf, 2019-07-01:
Following, looking for something similar.

felipecode, 2019-08-14:
I am working on something for that I will update you guys soon !

priyankanagaraj1494, 2019-09-17:
> Hi @Johnnyswong thank you very much, but I did not find any tutorial of how to build an RL environment at [https://github.com/openai/baselines](url). Can you give some more specific information?
> Thanks a lot!

hello,
refer this book "HANDS ON INTELLIGENT AGENTS WITH OPENAI GYM", they have clearly explained how to make CARLA as a new environment in openai gym. Also the codes are available here https://github.com/PacktPublishing/Hands-On-Intelligent-Agents-with-OpenAI-Gym/tree/master/ch7

zlw21gxy, 2019-11-14:
> I am working on something for that I will update you guys soon !

How things going on?

rajatsaini0294, 2019-12-04:
There are some tutorials on Youtube by the author named sentdex. You can check out.

closed 2020-02-09 by stale[bot]

rudrapoudel, 2020-07-03:
Any progress on this? Would really appreciate RL environment for CARLA 0.9.X. Thanks,

lijecaru, 2020-07-03:
Some environments compatible with Carla0.9.X https://github.com/Sentdex/Carla-RL, https://github.com/cjy1992/gym-carla, https://github.com/rail-berkeley/d4rl/tree/master/d4rl/carla

wu0023, 2021-12-30:
ThankS for sharing @lijecaru

SExpert12, 2024-06-26:
Anyone has new information regarding this? I am working on reinforcement learning in CARLA and I have tried many repositories but no one is working as code is written in old Tensorflow version or in other older version library. Please share if anyone has any tutorial link for reinforcement learning in CARLA simulator.