carla-simulator/carla #1783: libjpeg.so.8 PythonAPI dependency

qhaas, 2019-06-21:
We use CentOS 7 extensively at my site.  Both it and its upstream distribution (RHEL7) are widely used in the corporate and government world.  In some places, it is the only Linux operating system approved for use.

I am able to use the pre-built 0.9.5 binary distribution of the CARLA simulator (i.e. 'carla server') and UE4 provided by the CARLA developers with the libraries available in the mainstream CentOS 7 repositories.  My thanks for keeping this piece compatible with CentOS7.

However, when I try to use the CARLA PythonAPI (i.e. 'carla client') installed from the 'carla-0.9.5-py3.5-linux-x86_64.egg' provided with the 0.9.5 release, I find that 'libjpeg.so.8' is required by its native dependencies.  This library isn't available in the mainstream CentOS7 repositories.  The stable release of OpenSUSE provides an RPM containing it (seems to play nicely in CentOS7, but mixing packages meant for other distributions isn't ideal), one can build it from scratch, or one can (likely, though I have not tried) rebuild CARLA with a version of libjpeg available in the CentOS7 repositories.

Given this is the only dependency not easily available in the RHEL7 world (that I've discovered so far), I'd like to propose that the developers build against a version of libjpeg that is available in the mainstream repositories of the CentOS/RHEL7 and Debian/Ubuntu ecosystems, which should cover the vast majority of the Linux user base.  For example, 'libjpeg.so.62' is available in the libjpeg-turbo package in CentOS7, the libjpeg62 package in Ubuntu 16.04, and the libjpeg62-turbo package of Debian 9 (stable).

Thanks

nsubiron, 2019-06-26:
Hi @qhaas,

Yes, sounds reasonable, would you mind doing a pull request with the changes?

closed 2019-09-01 by stale[bot]

eerkaijun, 2021-02-22:
May I ask what is the current workaround for this issue? I tried to import carla but I also meet the problem of libjpeg.so.8 import error.