carla-simulator/carla #1790: clang 7 is required, but it's not installed.

yumianhuli2, 2019-06-21:
When I use ./Setup.sh under /carala/Util/BuildTools folder,
the command say clang 7 is required, but it's not installed 
How can I do next?

WolfgangFahl, 2019-07-20:
same here for make launch on Mac OS 10.13.6. Installed clang 7 with 

sudo port install clang-7.0 

to no avail. At least the message is misleading. I know there #150 but there seems to be no change in the main branch to make any steps in getting to a working build

cassfalg, 2019-07-22:
@yumianhuli2 can you please post more information like what system you are on, what steps you followed, what progress you made and so on?

WolfgangFahl, 2019-07-22:
You'll find quite a bit of information at http://wiki.bitplan.com/index.php/Car_simulation_software#Carla - there were so many things to tinker with and I tried on three different computers - there is IMHO just a lack of consistency in the support for different environments and operating systems. It would be good to have a script that first checks for the basics and tells what is missing before running into all the detail issues.

cassfalg, 2019-07-22:
@WolfgangFahl: Yes, that may be. I am also highly confused right now as to which part of the bundle is being built with which compiler, why they compile their own clang7 when they require a system clang7 and so on. But from what I gather Mac is not well supported, and I don't see any information from @yumianhuli2 that he is using a Mac. Are you and @yumianhuli2 the same guy? Or just someone with the same problem?

WolfgangFahl, 2019-07-22:
just same problem

asharalam11, 2019-07-22:
Issue #1878 might be of help

closed 2019-09-27 by stale[bot]