carla-simulator/carla #1791: sensor.other.obstacle

kimiya66, 2019-06-21:
Has anyone worked with this sensor? I want to get other cars and distance from vehicle to them. the problem is that when I use this distance attribute in sensor.other.obstacle to find distances to detected obstacle but it is really weird that its value is always 50 :
       	print('distance to obstacle:', data.distance)
                         output is:('distance to obstacle:', 50.0)   
during the simulation is always 50. 
I do not know how can I work with that. can any one please helping me how can I get information from nearby cars during simulation?

Thank you in advance,

kimiya66, 2019-06-22:
@nsubiron

kimiya66, 2019-06-23:
@JohnEmad92

closed 2019-08-29 by stale[bot]

cokeyke, 2020-05-18:
Same problem