carla-simulator/carla #1792: PythonAPI Build Problem in ARM64 Machine

ferhatmelih, 2019-06-21:
Thanks for creating easy to install python package via pip!
When I insall PythonAPI with command below
 >  pip install carla==0.9.5
in Arm64 Linux machine, I get the following error.

`
Collecting carla
  Using cached https://files.pythonhosted.org/packages/2e/16/c7ccc44d959a371e776c21644362d0716f7ef8244dcf1950912e131c41aa/carla-0.9.5.tar.gz
Building wheels for collected packages: carla
  Building wheel for carla (setup.py) ... error
  ERROR: Complete output from command /home/nvidia/Desktop/CarlaAPITest/carla-env/bin/python2 -u -c 'import setuptools, tokenize;__file__='"'"'/tmp/pip-install-7iRCU3/carla/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-9rGwC2 --python-tag cp27:
  ERROR: compiling:
    - source/libcarla/libcarla.cpp
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-aarch64-2.7
  creating build/lib.linux-aarch64-2.7/carla
  copying source/carla/__init__.py -> build/lib.linux-aarch64-2.7/carla
  copying source/carla/command.py -> build/lib.linux-aarch64-2.7/carla
  running egg_info
  writing source/carla.egg-info/PKG-INFO
  writing top-level names to source/carla.egg-info/top_level.txt
  writing dependency_links to source/carla.egg-info/dependency_links.txt
  reading manifest file 'source/carla.egg-info/SOURCES.txt'
  writing manifest file 'source/carla.egg-info/SOURCES.txt'
  running build_ext
  building 'carla.libcarla' extension
  creating build/temp.linux-aarch64-2.7
  creating build/temp.linux-aarch64-2.7/source
  creating build/temp.linux-aarch64-2.7/source/libcarla
  aarch64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -Wdate-time -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -fPIC -Idependencies/include -I/usr/lib/gcc/x86_64-linux-gnu/7/include -I/usr/include/python2.7 -c source/libcarla/libcarla.cpp -o build/temp.linux-aarch64-2.7/source/libcarla/libcarla.o -fPIC -std=c++14 -Wno-missing-braces -DBOOST_ERROR_CODE_HEADER_ONLY -DLIBCARLA_WITH_PYTHON_SUPPORT -DLIBCARLA_IMAGE_WITH_PNG_SUPPORT=true
  cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++
  source/libcarla/libcarla.cpp:7:26: fatal error: carla/Memory.h: No such file or directory
  compilation terminated.
  error: command 'aarch64-linux-gnu-gcc' failed with exit status 1
  ----------------------------------------
  ERROR: Failed building wheel for carla
  Running setup.py clean for carla
Failed to build carla
Installing collected packages: carla
  Running setup.py install for carla ... error
    ERROR: Complete output from command /home/nvidia/Desktop/CarlaAPITest/carla-env/bin/python2 -u -c 'import setuptools, tokenize;__file__='"'"'/tmp/pip-install-7iRCU3/carla/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-9jBhnG/install-record.txt --single-version-externally-managed --compile --install-headers /home/nvidia/Desktop/CarlaAPITest/carla-env/include/site/python2.7/carla:
    ERROR: compiling:
      - source/libcarla/libcarla.cpp
    running install
    running build
    running build_py
    creating build
    creating build/lib.linux-aarch64-2.7
    creating build/lib.linux-aarch64-2.7/carla
    copying source/carla/__init__.py -> build/lib.linux-aarch64-2.7/carla
    copying source/carla/command.py -> build/lib.linux-aarch64-2.7/carla
    running egg_info
    writing source/carla.egg-info/PKG-INFO
    writing top-level names to source/carla.egg-info/top_level.txt
    writing dependency_links to source/carla.egg-info/dependency_links.txt
    reading manifest file 'source/carla.egg-info/SOURCES.txt'
    writing manifest file 'source/carla.egg-info/SOURCES.txt'
    running build_ext
    building 'carla.libcarla' extension
    creating build/temp.linux-aarch64-2.7
    creating build/temp.linux-aarch64-2.7/source
    creating build/temp.linux-aarch64-2.7/source/libcarla
    aarch64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -Wdate-time -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -fPIC -Idependencies/include -I/usr/lib/gcc/x86_64-linux-gnu/7/include -I/usr/include/python2.7 -c source/libcarla/libcarla.cpp -o build/temp.linux-aarch64-2.7/source/libcarla/libcarla.o -fPIC -std=c++14 -Wno-missing-braces -DBOOST_ERROR_CODE_HEADER_ONLY -DLIBCARLA_WITH_PYTHON_SUPPORT -DLIBCARLA_IMAGE_WITH_PNG_SUPPORT=true
    cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++
    source/libcarla/libcarla.cpp:7:26: fatal error: carla/Memory.h: No such file or directory
    compilation terminated.
    error: command 'aarch64-linux-gnu-gcc' failed with exit status 1
    ----------------------------------------
ERROR: Command "/home/nvidia/Desktop/CarlaAPITest/carla-env/bin/python2 -u -c 'import setuptools, tokenize;__file__='"'"'/tmp/pip-install-7iRCU3/carla/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-9jBhnG/install-record.txt --single-version-externally-managed --compile --install-headers /home/nvidia/Desktop/CarlaAPITest/carla-env/include/site/python2.7/carla" failed with error code 1 in /tmp/pip-install-7iRCU3/carla/
`

Any help is appreciated.

praveen-palanisamy, 2019-06-25:
Hey @ferhatmelih , the PyPi repo for the carla pip package does not have pre-built binaries for ARM/ARM-HF architectures. The source build (which seems to have failed based on the error log you posted) is not supported at the moment.

I am not sure if CARLA/UE4 can run without issues on ARM arch that too with a mobile GPU. Your best bet is to use x86_64/amd64 based hardware platform (with a discrete GPU) for which, I have made sure that the carla pip (binary) package works on most of the systems.

closed 2019-09-03 by stale[bot]

hhuysqt, 2021-01-06:
Hi @ferhatmelih , I'm using python2.7 on an NVIDIA TX2 to connect to carla 0.9.11 that runs on my desktop server. I find out that I should compile the client part of carla from source code. As referenced [here](https://github.com/carla-simulator/carla/blob/master/Docs/build_linux.md#make-carla), this can be done by `make PythonAPI ARGS="--python-version=2"` which produces the `carla-0.9.11-py2.7-linux-aarch64.egg`. I add this egg file to `PYTHONPATH`, and successfully run the example python scripts. If you're using python3, just `make PythonAPI`.

vihar4u2005, 2021-01-27:
> Hi @ferhatmelih , I'm using python2.7 on an NVIDIA TX2 to connect to carla 0.9.11 that runs on my desktop server. I find out that I should compile the client part of carla from source code. As [...]

Hi @ferhatmelih I am trying to connect to carla which is on my windows pc to python3 from my tx2 on ubuntu.do i need to install carla on linux for this.

Hakistr, 2023-08-29:
Hello,
Is there any prebuilt Python libraries(.egg files etc.) that you can share for aarch64 platform?

ChgygLin, 2024-09-11:
@Hakistr I compiled 0.9.15 at RK3588, Ubuntu22.04 with clang10, g++12, python10. Hope it helps!
[https://drive.google.com/file/d/1BJM_ny5Kj5Iv9cc_oXsoYEG7bon7omtR/view?usp=drive_link](carla-0.9.15-cp310-cp310-linux_aarch64.whl) carla-0.9.15-cp310-cp310-linux_aarch64.whl

ChgygLin, 2024-09-29:
> @ChgygLin could you please reshare the link? can't seem to be able access it

Updated.

ChgygLin, 2024-09-29:
> @ChgygLin could you please reshare the link? can't seem to be able access it

@ashwin-999 Updated.

mohamedelouali1-creator, 2026-10-06:
I have cross compiled the Carla 0.9.13 Python API for python3.8 ubuntu 20.04 for ARM architecture aarch64, i hope it can be useful for the community : https://github.com/mohamedelouali1-creator/carla-python-api-aarch64-wheels/releases/tag/v0.9.13-aarch64