carla-simulator/carla #1793: how to see the definition of the Python API functions

zwbai, 2019-06-21:
Hi,

is there possible to see the definition of all the python API functions? In order to use these API functions, I need to know more about how it works internally and what is the return value. So can somebody tell me how to see that? 

Thank you very much!

Johnnyswong, 2019-06-24:
Carla_home/PythonAPI/agents

remitoudic, 2019-06-25:
Hey Zwbai , 
About the "python API functions"  , I didn't find anywhere proper definitions . As far I understand  in Carla_home/PythonAPI/agents you will find some  python classes build on the API  .  So I guess at the moment we have  to learn it  by trying and reading what has been already done  :)

nsubiron, 2019-06-25:
Most of the Python API methods are actually implemented in C++ in [LibCarla/source/carla/client](https://github.com/carla-simulator/carla/tree/master/LibCarla/source/carla/client), then exposed to Python using the bindings in [PythonAPI/carla/source/libcarla](https://github.com/carla-simulator/carla/tree/master/PythonAPI/carla/source/libcarla).

zwbai, 2019-06-25:
Thank you very much @nsubiron , here comes to my next question: can I modify the python API and rebuild them? I mean if I want to modify some API can I just change the code in LibCarla/source/carla/client and PythonAPI/carla/source/libcarla and then rebuild them? Thanks a lot! :D

nsubiron, 2019-06-25:
@zwbai of course, run `make PythonAPI` in the root folder to rebuild the Python module.

zwbai, 2019-06-25:
@nsubiron thanks a lot, maybe I have too many questions. but I really want to know how to get the light information of a certain traffic light even when the light is green. the `vehicle.get_traffic_light_state()` will only return the light information when the `vehicle.is_at_traffic_light()` is activated, which is not activated when the light is green. similar question with [issue#1799](https://github.com/carla-simulator/carla/issues/1799), but I really want to know that it is important for my project now. Thank you very much!

nsubiron, 2019-06-26:
@zwbai commented in #1799.

zwbai, 2019-06-26:
@nsubiron Thank you very much! Maybe some question in the future and this issue is closed now.

closed 2019-06-26 by zwbai