carla-simulator/carla #1795: Timeout in wait_for_tick(): Suspected Deadlock

ykarmesh, 2019-06-22:
Hi @felipecode & @nsubiron ,

I am getting errors related to timeouts from the function wait_for_tick() randomly when the code runs. This function is associated with either the scenario_runner classes CarlaActorPool and BasicScenario or it is called from my code. The exact error is -

RuntimeError: time-out of 60000ms while waiting for the simulator, make sure the simulator is ready and connected to localhost:2000
(I gave 60 seconds as the input to wait_for_tick() to see if the error goes away but it doesn’t)

I also get an exception while using load_world()  a couple of times. The error is - 

“failed to connect to newly created map" 
(The server is up and running at this point)

Both of these functions have a function called WaitForState() in common under the hood in the Carla library. The frequency of the timeout increases as I increase the number of client/servers or if other computationally expensive statements are added to the code. 
Also, something to note is that whenever the code gets stuck I see a few CPU cores’ (1 core per Carla) usage go to 100%. I suspect a deadlock occurring somewhere, which basically hangs the whole Carla server.

Specs of the machines:
CPU - Intel(R) Core(TM) i7-5930K CPU @ 3.50GHz (6 cores)
RAM - 62.8 GB
GPU - Quadro P6000 (24 GB)
OS - Ubuntu 18.04

Number of Clients/Server - 1-4
Mode for running carla - Built (Version cf58f7a)

Attached is the gist of the self-contained code that is a stripped down version of the actual code I’m using and getting errors in. It runs an agent in autonomous mode from a start point to a goal point while collecting data. I am using Scenario Runner to monitor the status of the agent in the environment so that I can terminate the episode once the agent’s status changes to SUCCESS or FAILURE. Apart from that, I am using the depth, segmentation and RGB camera. Removing the scenario runner from this code does reduce the number of occurrences of this error but it still occurs.

ykarmesh, 2019-06-22:
Edit: Checked with version 9694ba4, still the error is there.

Jiankai-Sun, 2019-06-24:
@nsubiron Same problem.

Johnnyswong, 2019-06-25:
@nsubiron Could you please take a look at this issue?

Johnnyswong, 2019-06-25:
@ykarmesh Hey, I've solved the problem. Just using the official carla container and run it with the host mode. My script is:

`docker run  --rm -d -it --net=host -e NVIDIA_VISIBLE_DEVICES=0 --runtime nvidia carlasim/carla:0.9.5 ./CarlaUE4.sh /Game/Carla/Maps/Town04 -carla-server -fps=20 -world-port=2000 -windowed -ResX=4 -ResY=4`

nsubiron, 2019-06-25:
I've been taking a look and I'm not 100% sure what's the issue, but I suspect it may be the race condition between the "tick" and "wait_for_tick" methods, basically doing this
```py
world.tick()
world.wait_for_tick()
```
it could happen that the tick arrives before we start waiting for it and we end up having a dead-lock. We have it like this in our sync-mode example, but I believe is wrong.

I've been testing, I'm going to make a PR with some changes; synchronizing the `tick` function to return the frame id of the frame it starts, and also changing the `apply_settings` so it returns the frame id at which the settings took effect. I think with that we can make a client that synchronizes safely with the simulator.

Give it a try and let me know if it fixes your problems!

ykarmesh, 2019-06-27:
Hi @nsubiron,

The solution works. There is no more deadlock. 

Thanks a lot!

closed 2019-06-27 by ykarmesh

Johnnyswong, 2019-06-30:
@ykarmesh I tried @nsubiron 's method. But failed with the error:

`RuntimeError: rpc::rpc_error during call in function version`
when I call the function of `client.get_world()`

ykarmesh, 2019-07-01:
Hi Johnny, 

I think this error occurs because you have different versions of the server and the client API. Can you try using the newer version of the API which comes with this PR. You should find the corresponding egg file inside the PythonAPI/carla/dist/ folder.

Johnnyswong, 2019-07-01:
Hi, @ykarmesh 
I'm using the 0.9.5 version of both simulator and client API. But my python version is 3.6.

Johnnyswong, 2019-07-01:
@ykarmesh And I've tried Python3.5, but it still couldn't work.

ykarmesh, 2019-07-01:
Try this:
`import carla`
 `client = carla.Client(host='localhost', port=port)`
 `server_version = client.get_server_version()`
 `client_version = client.get_client_version()`
Check if they are same or different