carla-simulator/carla #1796: Static Camera Sensor in Carla 0.9.5

Nozomeister, 2019-06-24:
Dear Carla team,
Is it possible to spawn a static camera sensor in Carla 0.9.5 without attaching it to the vehicle and save the image from that camera to disk? This feature is not possible in stable version.
Best regards,

kouyoumin, 2019-06-26:
Yes

Nozomeister, 2019-06-26:
> Yes

Can you suggest me one way to do it? Thank you so much

kouyoumin, 2019-06-26:
Try this:
```
import carla
from carla import ColorConverter as cc

def process_img(image):
    image.convert(cc.Raw)
    image.save_to_disk('_out/%08d' % image.frame_number)

client = carla.Client('127.0.0.1', 2000)
client.set_timeout(2.0)

world = client.get_world()
spectator = world.get_actors().filter('spectator')[0]

bp_library = world.get_blueprint_library()
camera_bp = bp_library.find('sensor.camera.rgb')
camera_bp.set_attribute('image_size_x', '640')
camera_bp.set_attribute('image_size_y', '360')

camera = world.spawn_actor(camera_bp, spectator.get_transform())
camera.listen(process_img)

try:
    while True:
        world.wait_for_tick()
except KeyboardInterrupt:
    camera.destroy()
    print('Exit')
```

sunwell1994, 2019-07-31:
This depends on spectator's view but spectator may move somehow. I want to deploy cameras at intersection automatically in a large scale. I use traffic light as anchor point to set up cameras but the view is very strange. Is there any advice?

DatKartoffel, 2019-08-05:
> This depends on spectator's view but spectator may move somehow. I want to deploy cameras at intersection automatically in a large scale. I use traffic light as anchor point to set up cameras but the view is very strange. Is there any advice?

Hey man, how can you attach the camera to the traffic light? I also want to do the same thing but it just doesn't work. Can you share some code?

sunwell1994, 2019-08-06:
> This depends on spectator's view but spectator may move somehow. I want to deploy cameras at intersection automatically in a large scale. I use traffic light as anchor point to set up cameras but [...]

It is not a perfect solution.
use `world.get_actors().filter()` to get traffic light list. You can attach the cameras to one traffic light but the view is terrible since the system gives a wrong value of `z` of the light(a value near to 0). Thus, you have to setup a relative `z`( I set it about 4.7) . Besides, you also need to calibrate the rotation for each camera to get a good view.

closed 2019-10-12 by stale[bot]

Harsharma2308, 2020-01-25:
If someone developed on mounting camera on traffic light, could you please share the script/tell me how to do it. Thanks

vignesharu, 2022-08-24:
> This depends on spectator's view but spectator may move somehow. I want to deploy cameras at intersection automatically in a large scale. I use traffic light as anchor point to set up cameras but the view is very strange. Is there any advice?

> This issue has been automatically marked as stale because it has not had recent activity. It will be closed if no further activity occurs. Thank you for your contributions.

import glob
import os
import sys
from typing_extensions import Self

try:
    sys.path.append(glob.glob('../carla/dist/carla-*%d.%d-%s.egg' % (
        sys.version_info.major,
        sys.version_info.minor,
        'win-amd64' if os.name == 'nt' else 'linux-x86_64'))[0])
except IndexError:
    print("carla error")
    pass

import carla
from carla import ColorConverter as cc
import random

def process_img(image):
    image.convert(cc.Raw)
    image.save_to_disk('_out/%08d' % image.frame_number)

# Connect to the client and retrieve the world object
client = carla.Client('localhost', 2000)
world = client.get_world() 

spawnpoint = carla.Transform(carla.Location(x=-19.350170,y=5.975828,z=5.672815),carla.Rotation(pitch=-16.132141,yaw=154.282227,roll=0.000017))

# We create the camera through a blueprint that defines its properties
camera_bp = world.get_blueprint_library().find('sensor.camera.rgb')
camera_bp.set_attribute('image_size_x', '1920')
camera_bp.set_attribute('image_size_y', '1020')

camera = world.spawn_actor(camera_bp, spawnpoint)

# actor_list.append(camera)
print('Working')
camera.listen(process_img)

try:
    while True:
        world.wait_for_tick()
except KeyboardInterrupt:
    camera.destroy()
    pass
finally:
    print('Exit')