carla-simulator/carla #1798: how to avoid Error  : " WARNING: sensor object went out of the scope but the sensor"

remitoudic, 2019-06-24:
Hey , 
I am a  brand new Carla starter...   I use  Carla (0.9.5) and Ubuntu 18. 
Can you help me to understand , what I did wrong or how to avoid this error: 

"WARNING: sensor object went out of the scope but the sensor" 

I want to  retrieve sensor Data . 

 my code is like that : 
` # set sensor `
` camera_bp = blueprint_library.find("sensor.camera.depth") ` 
 `  transform = carla.Transform(carla.Location(x= location.x , z= location.z)) `
 `sensor = world.spawn_actor(camera_bp, transform, attach_to=vehicle)`
 ` vehicle.set_autopilot(True)  `
` # retrieve data  `
`sensor.listen(lambda image: image.save_to_disk('TEST.png' % image.raw_data))`

If you have a clue , a hint or something to help me I will be very grateful.  I am very struggling at the moment.... 
Thank you in advance

nsubiron, 2019-06-25:
Hi @remitoudic,

You should destroy the sensor object when you're done with it so it does not remain in the simulation
```py
sensor.destroy()
```

remitoudic, 2019-06-25:
Thank you  very   much ! 

I get rid of the error thank to your help.  Actually I totally missed it  in  the ....PythonAPI/example/tutorial.py  file maybe because it was at the end  of the code.   In addition I also  did not  understand in the C++ code   
`if (IsAlive() && IsListening()) {
     log_warning(
         "sensor object went out of the scope but the sensor is still alive",`
what I should do to not having the sensor alive and not listening if I want to retrieve data .

closed 2019-06-25 by remitoudic

varunjammula, 2019-08-13:
Same issue!

YuanYunshuang, 2020-08-25:
I also met this error, and finally figure out what is happening. I assigned the spawned sensor to a local variable and then set the sensor object to listen(). Then I only stored the sensor actor id in the attribute of a customer object. So this would cause a problem because of the memory releasing once your code finishes you function, in which you spawned the sensor. This is also why the error tells you the sensor object is out of scope. take your code snippet as an example:

If your code is implemented under a function scope or a membership function of an object like this:

`sensor_list = []

def func():
   # set sensor
   camera_bp = blueprint_library.find("sensor.camera.depth")
   transform = carla.Transform(carla.Location(x= location.x , z= location.z))
   sensor = world.spawn_actor(camera_bp, transform, attach_to=vehicle)
   vehicle.set_autopilot(True)
   # retrieve data
   sensor.listen(lambda image: image.save_to_disk('TEST.png' % image.raw_data))
   sensor_list.append(sensor.actor_id) # or even not store the id would also lead to the error once it finishes this func.
`
To avoid this warning, I only changed 
`sensor_list.append(sensor.actor_id)` to `sensor_list.append(sensor)`

I hope this could help :-)

tinmodeHuang, 2020-10-20:
I would appreciate it if someone gives me a hand!
Q1. why can I still feel lagging obviously in between ticks (or frames) in spite of set the time as 0.1 seconds? and how shall I do?
Q2. why is there the WARNING in the console, and how to solve it?

code blocks about the two problems as follows:
```python
class Env():
    im_width = 640
    im_height = 480
    front_camera = None

    def __init__(self, args):
        rgb_cam = self.world.get_blueprint_library().find('sensor.camera.rgb')
        self.sensors = []
        self.sensors.append(rgb_cam)
        self.collision_sensor = self.world.get_blueprint_library().find("sensor.other.collision")
        self.sensors.append(self.collision_sensor)

    def reset(self):
        try:
            #--------------
            # Spawn sensors
            #--------------
            # spawn the collision detector in the end in order to avoid detecting the collision
            # caused by the ego vehicle falling and having impact on other sensors working normally.
            index = self.sensors.index(self.collision_sensor)
            collision_detector = self.sensors.pop(index)
            self.sensors.append(collision_detector)

            transform_sensor = carla.Transform(carla.Location(x=2.5, z=0.7))
            for i in range(len(self.sensors)):
                if self.sensors[i].id == 'sensor.camera.rgb':
                    self.sensors[i].set_attribute("image_size_x", f"{self.im_width}")
                    self.sensors[i].set_attribute("image_size_y", f"{self.im_height}")
                    self.sensors[i].set_attribute("fov", f"110")
                    self.sensor = self.world.spawn_actor(self.sensors[i], transform_sensor, ego_vehicle)
                    sensors_list.append(self.sensor)
                    self.sensor.listen(lambda data: self.process_img(data))
                elif self.sensors[i].id == 'sensor.other.collision':
                    self.sensor = self.world.spawn_actor(self.sensors[i], transform_sensor, ego_vehicle)
                    sensors_list.append(self.sensor)
                    self.sensor.listen(lambda data: self.collision_data(data))
            while True:
                # if args.sync and self.synchronous_master:
                #     self.world.tick()
                # else:
                #     self.world.wait_for_tick()
                self.world.tick()
            return self.front_camera
        finally:

            if args.sync and self.synchronous_master:
                settings = self.world.get_settings()
                settings.synchronous_mode = False
                settings.fixed_delta_seconds = None
                self.world.apply_settings(settings)

            print('\ndestroying %d vehicles' % len(vehicles_list))
            self.client.apply_batch([carla.command.DestroyActor(x) for x in vehicles_list])

            print('\ndestroying %d sensors' % len(sensors_list))
            self.client.apply_batch([carla.command.DestroyActor(x) for x in sensors_list])

            # stop walker controllers (list is [controller, actor, controller, actor ...])
            for i in range(0, len(all_id), 2):
                all_actors[i].stop()

            print('\ndestroying %d walkers' % len(walkers_list))
            self.client.apply_batch([carla.command.DestroyActor(x) for x in all_id])
```
outputs as follows:
```bash
destroying 265 vehicles

destroying 2 sensors

destroying 70 walkers

done.
Traceback (most recent call last):
  File "spawn_actor.py", line 327, in <module>
    env.reset()
  File "spawn_actor.py", line 270, in reset
    self.world.tick(0.1)
RuntimeError: time-out of 10000ms while waiting for the simulator, make sure the simulator is ready and connected to localhost:2000
WARNING: sensor object went out of the scope but the sensor is still alive in the simulation: Actor 6466 (sensor.camera.rgb)
```
@YuanYunshuang

jewes, 2021-06-11:
call sensor.stop() if sensor.listen() was called before exiting.

XdBirdie, 2021-07-24:
> I also met this error, and finally figure out what is happening. I assigned the spawned sensor to a local variable and then set the sensor object to listen(). Then I only stored the sensor actor id in the attribute of a customer object. So this would cause a problem because of the memory releasing once your code finishes you function, in which you spawned the sensor. This is also why the error tells you the sensor object is out of scope. take your code snippet as an example:
> 
> If your code is implemented under a function scope or a membership function of an object like this:
> `sensor_list = []`
> 
> `def func():`
> ` camera_bp = blueprint_library.find("sensor.camera.depth")`
> ` transform = carla.Transform(carla.Location(x= location.x , z= location.z))`
> ` sensor = world.spawn_actor(camera_bp, transform, attach_to=vehicle)`
> ` vehicle.set_autopilot(True)`
> ` sensor.listen(lambda image: image.save_to_disk('TEST.png' % image.raw_data))`
> ` sensor_list.append(sensor.actor_id) # or even not store the id would also lead to the error once it finishes this func.`
> 
> To avoid this warning, I only changed
> `sensor_list.append(sensor.actor_id)` to `sensor_list.append(sensor)`
> 
> I hope this could help :-)

It works and helps me a lot !!!  I'm going crazy due to the problem. Thank you very much!!