carla-simulator/carla #1800: lidar sensor

kimiya66, 2019-06-24:
If I understand correctly, raw_data as one of the attr of lidar's data gives me the location of detected obstacle? however, it gives a buffer which seems that contains pointers but how can I get the location of detected vehicles from this buffer. how can open this buffer. 
could any one please help me? I would be really grateful.
Thank you
@nsubiron

jjarquin, 2019-08-16:
Hi, the lidar sensor doesn't detect obstacles, it shoots rays and provides you a point where the laser hits a surface.

If you are in python you can easily convert it to a numpy array

````
lidar_data = np.frombuffer(raw_data, dtype=np.float32).reshape([-1, 3])
````

closed 2019-10-22 by stale[bot]

FrancescoTerrosi, 2019-12-02:
Hello, what's the interpretation given to the point_cloud array?
I mean, what does the element in position x-y represent with respect to the car? ty