carla-simulator/carla #1806: 0.9.5 Register new sensor and publish via carla_ros_bridge

slyer0009, 2019-06-26:
Hi everybody! 

I'm trying to add a new "own" sensor to Carla. What I've done is following [these steps](https://github.com/carla-simulator/carla/pull/830#issue-219758676) to register a new sensor. I just duplicated the Lidar sensor and renamed it and all its dependencies (including the files in the CarlaUE4 Plugins folder). It worked as far as I'm able to spawn the sensor and attach it to a vehicle.

Now up to the next step, the ROS-Bridge. I duplicated the lidar.py here aswell and renamed it properly.But when I try using the ROS-Bridge with the new sensor I get the following error:

`terminate called after throwing an instance of 'clmdep_msgpack::v1::type_error'
  what():  std::bad_cast`

I believe that the sensor is unknown for the bridge or something? Maybe I missed a step registering the new sensor somewhere else? 

Any ideas are welcome! 

Kind regards, 
Slyer

closed 2019-09-01 by stale[bot]

CLEMENTJOHNSHAJI, 2020-05-12:
Hi @slyer0009, Did you get around with this issue?

Lucbus, 2020-12-10:
Are there any updates on this? How did you resolve this?