carla-simulator/carla #1808: Build CarlaUE4 from source,make launch freezed.

jahad9819jjj, 2019-06-26:
Hi, I appreciate you to create Carla!
I try to build carla 0.9.x on Ubuntu16.0.4 by using UE4.22.

I follow the document and can run correctly.
but I did `make launch` and receive a below error.
Then launching CarlaUE4 freezed in the middle.

```
Error: CDO Constructor (CarlaMapGenerator): Failed to find StaticMesh'/Game/Carla/Static/RoadLines/St_Road_XCross_LaneMarking.St_Road_XCross_LaneMarking'
 Error: CDO Constructor (CarlaMapGenerator): Failed to find StaticMesh'/Game/Carla/Static/SideWalk/St_Road_XCross_Sidewalk4.St_Road_XCross_Sidewalk4'

```

closed 2019-06-29 by jahad9819jjj

BenMSK, 2019-08-27:
I got similar errors but solved it. 
I think that the problem is caused by, mentioned files in 'Contents folder' don't exist.

The following commands for solving are, 
1) At the cloned folder, type "$make clean". It deletes all dependencies of UE4.22.
2) "$git pull" (if your version of Carla is not up-to-date.
3) "$./Update.sh" --> It will update all needed files into 'Contents folder'.
4) "make launch" again.

I am not sure that I solved it correctly, but it worked :)

CQUlifuling, 2021-12-21:
hello, how did you finally solve this problem?