carla-simulator/carla #1809: Tick function for Reinforcement Learning

kraken24, 2019-06-26:
Hello Everyone, 

I am using Carla 0.9.4 for developing reinforcement learning algorithm for autonomous driving.

I have 2 queries:
1. I am confused between world.tick, world.on_tick and world.wait_for_tick functions. I do not understand which function to use when?

2. I want simulator to perform certain step only when I give signals and it should stop when I train my neural network. 

I have checked synchronous_mode.py but i still get error that: failed to connect to newly created map

nsubiron, 2019-06-26:
Hi @kraken24,

First of all, 0.9.4 has several issues with synchronous mode that have already being addressed in master, among them dead-locks when changing the map. Also take a look at #1803 as it may have some further fixes.

The thing you need to know is that every time the simulator does an update (advances the simulation one step), it sends a message to the clients that are connected. We also call this update a "tick".

* `world.wait_for_tick()` blocks until a new tick is received from the simulator. This function is not very reliable as you may skip ticks, and you can easily end up in a dead-lock if using synchronous mode.
* `world.on_tick(callback)` registers a callback (function or function object) to be called every time a new tick is received, the callback is executed asynchronously in the background.
* `world.tick()` tells the simulator to perform a tick (i.e. advance one step). This only has effect in synchronous mode.

So if you enable synchronous mode, the simulator is going to halt until you call `world.tick()`.

kraken24, 2019-06-27:
thank you @nsubiron 

Now i am able to work with synchronous mode.

But I have 1 more problem, after spawining the vehicle and setting a fix velocity as initial setting,
1. if I use set_velocity(vehicle.get_velocity()), it works perfectly and I get constant velocity of vehicle
2. if I use vehicle.apply_control(throttle=1) then vehicle decelerates few seconds and then accelerates.

is it because I use only 10 fps?

[image]

nsubiron, 2019-06-27:
`set_velocity(...)` bypasses the vehicle model, it just applies the velocity directly in the physics engine, so you get the net velocity you want. On the other hand, `apply_control(...)` uses the vehicle model to calculate the final velocity; i.e. takes into account gear shifting, engine RPM, etc. Probably the vehicle decelerates because is shifting gears.

kraken24, 2019-06-27:
Thanks @nsubiron 

Is there any documentation that explains relationship between carla vehicle control parameters and vehicle parameters. for example variation of rate of acceleration as i change throttle values. (all other parameters are kept constant)

nsubiron, 2019-06-28:
Not really, we just pass these values to the physics engine, [PhysX Vehicles](https://docs.nvidia.com/gameworks/content/gameworkslibrary/physx/guide/Manual/Vehicles.html).

closed 2019-09-03 by stale[bot]

priyankanagaraj1494, 2020-02-03:
> Hi @kraken24,
>
> First of all, 0.9.4 has several issues with synchronous mode that have already being addressed in master, among them dead-locks when changing the map. Also take a look at #1803 as it [...]

Hi @nsubiron ,

What is does the parameter seconds in the function world.wait_for_tick() mean?

shuhangchen, 2020-06-16:
> Hi @kraken24,
>
> First of all, 0.9.4 has several issues with synchronous mode that have already being addressed in master, among them dead-locks when changing the map. Also take a look at #1803 as it [...]

Hi @nsubiron,
 
       I tried to control the car in synchronous mode, but the car was not moving.
       For example, in the example/synchronous_mode.py file, the current program "controls" the car by modifying the transform of the car directly. However, the car woud not react (move) if I use something like 
>  vehicle.apply_control(carla.VehicleControl(throttle=1.0, steer=0.0))
     
Can I ask if I did something wrong? or how to make the control work in this case?

Thanks!