carla-simulator/carla #1816: distance of each vehicle's side to other obstacles

kimiya66, 2019-06-28:
what is the best way to get the distance of each vehicle's side to other obstacles?
As I saw with getLocation() method I could just get the central location of the actor in the world and I could not get the coordinate of different points of the vehicle.
I have also tried other.obstacle sensor but with distance attribute which is supposed to get distance to other obstacle, I always get a constant value of 50 which is weird. does this sensor actually work?
@dnovillo-CVC @nsubiron  @marcgpuig @iFuSiiOnzZ
 I would be thankful for your help

closed 2019-09-03 by stale[bot]

dkarunakaran, 2019-11-15:
@kimiya66 I am having the same problem. Did you solve this?

marcgpuig, 2019-12-10:
Did #2182 fixed this problem?