carla-simulator/carla #1817: Scenario Runner Runtime Error: failed to connect to newly created map

programmer6288, 2019-06-28:
Hello,

I am running the simulator on an Acer Aspire E15 laptop and have Windows 10.
When I try to run the scenario runner script, I get this error:

> D:\scenario_runner\scenario_runner-master>python scenario_runner.py --scenario FollowLeadingVehicle_1
Traceback (most recent call last):
  File "scenario_runner.py", line 287, in <module>
    SCENARIORUNNER.run(ARGUMENTS)
  File "scenario_runner.py", line 198, in run
    self.world = self.client.load_world(config.town)
RuntimeError: failed to connect to newly created map

The error says "failed to connect to newly created map". Does this mean that it could not connect to the simulation? I know there are different errors (like timeout 10000ms) when the script cannot connect to the simulation, so I don't think that is what it is.

If anybody has had or can recreate this error, it would be a big help

ykarmesh, 2019-07-01:
Hi, 

If you refer to this line in Carla's code [[Link]](https://github.com/carla-simulator/carla/blob/cfabe549c79b7c58e21d6136abd2cb45b868b038/LibCarla/source/carla/client/detail/Simulator.cpp#L71), you should see that the code waits 10 times for 2 secs to see if the episode was loaded. If its not loaded the clients gives you an error. Sometimes I saw that the episode would take 20+ secs to load and I would get an error by then. I would request @nsubiron to provide this as a function argument to pass with load_world().

BrandonHaynes, 2019-07-01:
I've seen this behavior before as well, and rerunning usually works around the problem.

Rather than adding a new parameter to `load_world`, perhaps we could just (approximately) respect the client timeout parameter?

programmer6288, 2019-07-01:
> Hi,
> 
> If you refer to this line in Carla's code [[Link]](https://github.com/carla-simulator/carla/blob/cfabe549c79b7c58e21d6136abd2cb45b868b038/LibCarla/source/carla/client/detail/Simulator.cpp#L71), you should see that the code waits 10 times for 2 secs to see if the episode was loaded. If its not loaded the clients gives you an error. Sometimes I saw that the episode would take 20+ secs to load and I would get an error by then. I would request @nsubiron to provide the timeout as a function argument to pass with load_world().

Hello @ykarmesh,
Thank you for showing me where the error is thrown in the code. However, I am not sure what you mean by your solution. I assume that you have encountered this problem before. What did you do to solve it? If the function load_world() does not take the timeout as an argument, how would it be changed to allow it as an argument?

Thank you

programmer6288, 2019-07-01:
> I've seen this behavior before as well, and rerunning usually works around the problem.
> 
> Rather than adding a new parameter to `load_world`, perhaps we could just (approximately) respect the client timeout parameter?

Hello @BrandonHaynes,
I have rerun several times, and I get the same error. The new map does load, but the script fails to connect. 

Also, what do you mean by "respect the client timeout parameter"?

Thank you

BrandonHaynes, 2019-07-01:
Ideally, this code should also modify the `load_world` timeout (*but does not currently do so*):

```
client.set_timeout(999)
client.load_world(...)
```

In the meantime, you can always just increase the timeout in the code that @programmer6288 linked and rebuild :)

ykarmesh, 2019-07-01:
Hi @programmer6288,

I don't have a solution to it apart from using try and except blocks to catch the error. Then I just try using load_world 3 more times before stopping the program. However, this is just a hack and not the best solution. So I believe that increasing the timeout in the main code might be the best idea. @nsubiron could tell you better.

programmer6288, 2019-07-02:
Hello @ykarmesh,

I don't seem to have the code that you linked to in your first post. I downloaded Carla using this link: https://github.com/carla-simulator/carla/releases. 

Do I need to install Carla differently to have access to the source code? I tried following the directions to build from source on windows, but I got lost somewhere.

Thank you

ykarmesh, 2019-07-02:
The only way is to build it from source. I dont have an idea about how to do it in windows.

cjerome0321, 2019-07-16:
Hello @programmer6288 ,
I met the same error, what I tried is to ignore this error, since as you said, the new map does load even though the script fails to connet:
```
try: 
    self.client.load_world(config.town)
except:
    pass
self.world = self.client.get_world()
```
However I met another error when I run scenario runner with reloadWorld on: 
```
File "scenario_runner.py", line 269, in run
    CarlaDataProvider.set_world(self.world)
File "E:\CARLA_0.9.5\scenario_runner-0.9.5.1\srunner\scenariomanager\carla_data_provider.py", line 160, in set_world
    CarlaDataProvider._map = CarlaDataProvider._world.get_map()
RuntimeError: trying to access an expired episode; a new episode was started in the simulation but an object tried accessing the old one.
```
I found a similar situation here, but it seems still unsolved yet:
[https://github.com/carla-simulator/scenario_runner/issues/291](url)

closed 2019-09-22 by stale[bot]