carla-simulator/carla #1818: LogarithmicDepth Image decode

NCTUyoung, 2019-06-29:
I have collect data using the examplefile `manual_control.py` 

Is there any way to decode the LogarithmicDepth png I get ?
The document seem to only mention about how to decode rgba png file.

I alse check the c++  source code of color convertor , only find `depth` and `LogarithmicLinear`

nsubiron, 2019-07-01:
Logarithmic depth is calculated applying those two: `LogarithmicLinear(Depth(pixel))`.

We recommend decoding the RGB version, it has more precision. Logarithmic depth is only meant for visualization.

NCTUyoung, 2019-07-03:
Got it!
I find there is `numpy.clip()` operation  when transferring to Logarithmic format.
Thanks for your kindly remind 👍

closed 2019-07-03 by NCTUyoung