carla-simulator/carla #1821: PythonAPI sensor actor cleanup "no stream available"

fpasch, 2019-07-01:
There seems to be an issue within the cleanup of the sensor actors, which results in the following error messages:
```
ERROR: Invalid session: no stream available with id <...>
...
```
This was seen within the CARLA ROS bridge. It seems to happen whenever a sensor, that has a [registered callback](https://github.com/carla-simulator/ros-bridge/blob/508f309c299f6f04691df05e89c5c46eb62b9b2e/carla_ros_bridge/src/carla_ros_bridge/sensor.py#L52) (`self.carla_actor.listen(self._callback_sensor_data)`) goes away.

From my knowledge there is no API available to handle that issue from within Python. (e.g. a callback whenever the actor goes away).

So I believe that we need to handle it within the CARLA PythonAPI code.

nsubiron, 2019-07-01:
It could be very handy having this callback, I think we can add a list of destroyed actors in the world observer's message, that'd make very easy adding callbacks in the client-side. This would be also very handy for the pedestrian navigation.

Meanwhile a workaround could be checking periodically if the actor is still alive
```py
if world.get_snapshot().find(actor_id) is None:
    # actor was destroyed.
```

fpasch, 2019-07-01:
At the moment there are two tick()-callbacks registered in CARLA ROS bridge:
1. update list of actors (this is done once per seconds) ([Code](https://github.com/carla-simulator/ros-bridge/blob/482be5fdbd92f6affc74e224f7ab7013c61cdd40/carla_ros_bridge/src/carla_ros_bridge/bridge.py#L113))
2. update time and current state of actors (on every tick) ([Code](https://github.com/carla-simulator/ros-bridge/blob/482be5fdbd92f6affc74e224f7ab7013c61cdd40/carla_ros_bridge/src/carla_ros_bridge/bridge.py#L90))

Therefore the error-message is only seen for max 1s.

So ideally, we would have two callbacks:
1. whenever a new actor was spawned
2. whenever an actor was deleted

germanros1987, 2020-06-26:
@fpasch Is this still a relevant issue?

fpasch, 2020-06-26:
Yes.

germanros1987, 2020-06-27:
Would it be fine to just keep a list with the deleted actors and offer that information to the client?

fpasch, 2020-06-29:
This issue has two aspects:
1. How to find deleted actors
2. Error message from CARLA Client Library

For 1.),  although it would be convenient to have a callback on actor-deletion, there is the workaround to compare the actor lists between ticks (see [ros-bridge](https://github.com/carla-simulator/ros-bridge/blob/master/carla_ros_bridge/src/carla_ros_bridge/bridge.py#L297)). So this has a low priority.
What should get investigated is 2.), the clean up of sensors with registered callback.

germanros1987, 2020-06-29:
Added to the task board.