carla-simulator/carla #1825: How to  avoid this Error   :  module 'carla' has no attribute 'WorldSnapshot'

remitoudic, 2019-07-01:
Hey  all of you, 

I use ubunutu 18.04 and carla 0.9.5 . 

As a carla starter starter I struggle a bite.  Can you help me with this issue: 

I would like to have access to the  carla.WorldSnapshot  or carla.World.get_snapshot()  but  I don't manage it . I  followed the documentation and  read issues #1776 and #1274   but unfortunately I  don't  know  what I am missing. 

With the code below I get the following Error :  **module 'carla' has no attribute 'WorldSnapshot'**

my code: 
... # setting 
...
`world = client.get_world()`
....  # I put a car in the simulation  and let it drive in autopilot 

 ... # want to retrieve a snapshot of the world at this point in time.
`world_snapshot = world.get_snapshot()`

... # Register callback to get called every time we receive a new snapshot.
`world.on_tick(lambda world_snapshot: print( 'debugging done !)) `

So ,  first reflex was to check if the  method/ attibute exist in carla / world  class  but I  did not find it either  by my self .    So where is it ?  or what I am missing ? 

Thank you in advance :)

nsubiron, 2019-07-01:
Carla 0.9.5 is previous to those changes, you'll need to get the [nightly build](http://carla-assets-internal.s3.amazonaws.com/Releases/Linux/Dev/CARLA_Latest.tar.gz) to have those.

remitoudic, 2019-07-01:
Ah ok it  make sense now ....
Thank you very much

closed 2019-07-01 by remitoudic