carla-simulator/carla #1827: Delay in transfer of controls from client to Carla Server while implementing imitation learning model

ritvik06, 2019-07-01:
I am implementing an imitation learning model to train the car to be trained on a CNN style model.
I load the model in my <b>client_example.py</b> using keras.models.load_model and then predict the control.steer values using prediction done by the model. The following is a code snippet from the client when the args values is autopilot

control = measurements.player_measurements.autopilot_control
img_array = np.asarray(img)
np.reshape(img_array, img_array.shape + (1,))
control.steer = float(model.predict(img_array[None,300:400,:,:], batch_size=1))

i run the following for the client - <i> python3.5 client_example.py --autopilot </i> and the server using <i> DISPLAY= ./CarlaUE4.sh -quality-level=Low -carla-server -benchmark -fps=10 </i>

I am getting my FPS = 3 and the simulation is running very slowly, the car is barely responding and the speed is 1km/h long after the connection is made on port 2000. Both the client and the server are being run on the same PC. I have deleted all the cameras(depth cameras too) and kept only the RGB image camera as input. The client server connection works perfectly with a higher frame rate when the autopilot argument is not given. Even when i do not predict the steer value using my model and let carla run it's own AI with args as autopilot, the frame rate and the speed is very low. The GPU i am using is a NVIDIA GeForce GTX 1080 with a memory space of about 11 GB. I have installed driver version 390 with cuda 9.2 and cudnn 7.6.0. 

Please help me out on how to fix the issue of the delay/ Slow speed of controls being and sent and the the variation in the running of the simulator because of the autopilot mode.

Derekabc, 2019-07-03:
Hello, Have you solved the problem?  Nowadays, I train the ego car based on DQN model. At the beging, the FPS is 20fps for both server and client, and the agent is trying to learn something as the score is better and better. While after hundreds of episode, the FPS of the cilent drops to 0 and the vehicle just stops at the start point.  I do not know what is the reason for this.

ritvik06, 2019-07-04:
No, i haven't. Waiting for a response.

closed 2019-07-15 by ritvik06

Micahmichael03, 2024-08-04:
any reply or have you seen the problem. am having the same problem. please let me know. doing this as my final year project in my university.