carla-simulator/carla #1834: How to get actor's longitudinal/lateral size and coordination.

kimiya66, 2019-07-02:
How can I get longitudinal and lateral size of vehicle? and also how to get coordinate of each point of the car? 
As I saw get_location just get the center of actor.

n1try, 2019-07-03:
For cars and pedestrians you can just read the `.bounding_box` property. For other actors, it seems like that's currently not supported (see #1832).

kimiya66, 2019-07-03:
bounding_box hast attributes location and extent. As I see location also gets the central location of box according to vehicle. however, I want the coordination for each point of the vehicle(ie, side of vehicle or back of the vehicle not just the central one)

n1try, 2019-07-03:
I guess you can derive them from the vehicle's position and the bounding box's extent. First, get vehicle's location (`actor.get_location()` -> return vehicle's center point), then get the BBox and than "add" / "subtract" (in Cartesian space) the extent to the center point.

n1try, 2019-07-03:
Take a look at this example: https://github.com/carla-simulator/carla/blob/master/PythonAPI/carla/scene_layout.py#L123

closed 2019-09-08 by stale[bot]