carla-simulator/carla #1835: Sensor falls down

francescoPolizzi, 2019-07-02:


francescoPolizzi, 2019-07-02:
hi, i have  attached a same sensors using the _"tutorial.py"_ file. The problem is the the sensors fall dawn into the street and doesn't remain attacched to the vehicle although i specified the _"attach_to=vehicle"_ argoument in the _spawn_actor_ function. As result appear the following Warning:
_WARNING: sensor object went out of the scope but the sensor is still alive in the simulation: Actor 104 (sensor.camera.rgb)_
here the code part of interest:
[image]

kimiya66, 2019-07-02:
I think the problem is because of camera_transform. carla.location should be said regards to vehicle and is not the location in the world. and x=-5.5 is out of the scope of vehicle actor. as it is the longitudinal distance from center of vehicle and 5.5 is a lot.

francescoPolizzi, 2019-07-03:
> I think the problem is because of camera_transform. carla.location should be set according to vehicle and is not the location in the world.  x=-5.5 is out of the scope of vehicle actor as it is the longitudinal distance from center of vehicle and 5.5 is a lot.
@kimiya66  thank you for the answer
Unfortunately i have tried also with x = 1.5 and z=2.4 and the result is the same. if do you have a portion of code used by you to do the same thing can you posts it here ? 
thanks again.

kimiya66, 2019-07-03:
I defined it as these and it was working: 

camera_blueprint = world.get_blueprint_library().find('sensor.camera.rgb')
camera_location = carla.Transform(carla.Location(x=1.2, z=1.7))
camera_blueprint.set_attribute('sensor_tick', '0.5');
camera_sensor = world.spawn_actor(camera_blueprint, camera_location, attach_to=vehicle);

francescoPolizzi, 2019-07-03:
Unfortunately there is still the same problem.
There all the code:

`
    try:
        # First of all, we need to create the client that will send the requests
        # to the simulator. Here we'll assume the simulator is accepting
        # requests in the localhost at port 2000.
        client = carla.Client('localhost', 2000)
        
        client.set_timeout(2.0)

        # Once we have a client we can retrieve the world that is currently
        # running.
        world = client.get_world()

        # The world contains the list blueprints that we can use for adding new
        # actors into the simulation.
        blueprint_library = world.get_blueprint_library()

        # Now let's filter all the blueprints of type 'vehicle' and choose one
        # at random.
        bp = random.choice(blueprint_library.filter('vehicle'))

        # A blueprint contains the list of attributes that define a vehicle's
        # instance, we can read them and modify some of them. For instance,
        # let's randomize its color.
        if bp.has_attribute('color'):
            color = random.choice(bp.get_attribute('color').recommended_values)
            bp.set_attribute('color', color)

        # Now we need to give an initial transform to the vehicle. We choose a
        # random transform from the list of recommended spawn points of the map.
        transform = random.choice(world.get_map().get_spawn_points())

        # So let's tell the world to spawn the vehicle.
        vehicle = world.spawn_actor(bp, transform)

        # It is important to note that the actors we create won't be destroyed
        # unless we call their "destroy" function. If we fail to call "destroy"
        # they will stay in the simulation even after we quit the Python script.
        # For that reason, we are storing all the actors we create so we can
        # destroy them afterwards.
        actor_list.append(vehicle)
        print('created %s' % vehicle.type_id)

       

        # Let's add now a "depth" camera attached to the vehicle. Note that the
        # transform we give here is now relative to the vehicle.
        camera_rgb = blueprint_library.find('sensor.camera.rgb')
        
        camera_rgb2 = blueprint_library.find('sensor.camera.depth')
        camera_rgb3 = blueprint_library.find('sensor.camera.semantic_segmentation')
        camera_rgb4 = blueprint_library.find('sensor.lidar.ray_cast')

        camera = carla.Transform(carla.Location(x=1.2, z=1.7))
        lidar =carla.Transform(carla.Location(x=3, y=0, z=0), carla.Rotation(roll=0,yaw=90,pitch=0))
        #camera = carla.Transform(carla.Location(x=0,y=5,z=5), carla.Rotation(roll=0,yaw=0,pitch=0))
        camera_tf_2 =carla.Transform(carla.Location(x=-5, z=5), carla.Rotation(roll=0,yaw=0,pitch=-45))

        camera1 = world.spawn_actor(camera_rgb, camera, attach_to=vehicle)
        #camera2 = world.spawn_actor(camera_rgb, camera_tf_2, attach_to=vehicle)
        #camera2 = world.spawn_actor(camera_rgb2, camera_transform, attach_to=vehicle)
        #camera3 = world.spawn_actor(camera_rgb3, camera_transform, attach_to=vehicle)
        #camera4 = world.spawn_actor(camera_rgb4, lidar, attach_to=vehicle)
        # Now we register the function that will be called each time the sensor
        # receives an image. In this example we are saving the image to disk
        # converting the pixels to gray-scale.
        #cc = carla.ColorConverter.LogarithmicDepth

        #COLOR CONVERTER
        #cc_segmentation = carla.ColorConverter.CityScapesPalette
        cc_depth= carla.ColorConverter.Depth
        
        

        print("listen")
        
        

        #camera1.listen(lambda image: to_bgra_array(image, 'Data8/RGB_',image.frame_number))
        #camera2.listen(lambda image: to_bgra_array(image,'Data8/DEPTH_',image.frame_number))
        #camera3.listen(lambda image: to_bgra_array(image,'Data8/SEGMENTATION_',image.frame_number))
        #camera4.listen(lambda image: image.save_to_disk('Data12/LIDAR_%06d' % image.frame_number))
        camera1.listen(lambda image: image.save_to_disk('Data12/RGB_%06d' % image.frame_number))
        #camera2.listen(lambda image: image.save_to_disk('Data12/RGB_camera2_%06d' % image.frame_number))
        #print(a)
        

           
           
        time.sleep(5)

    finally:

        
        print('done.')

if __name__ == '__main__':

    main()
`

francescoPolizzi, 2019-07-13:
up

closed 2019-09-18 by stale[bot]

Tanman1234, 2021-05-21:
> up

can you show for obstacle.detector sensor also i am getting same error with obstacle.detector