carla-simulator/carla #1839: How can I update my script this the latest Carla version

remitoudic, 2019-07-03:
Hey all of you , 
I use Ubuntu 18 and the  "nightly build " version of  Carla . ( I want to have access to the 
 carla.WorldSnapshot  feature that is why I want to use the latest version.)

My problem is that my previous scripts  for  Carla 0.9.5  are not working anymore and I don't have any working examples  or  documentation to update them.  Nearly nothing  works anymore  in  PythonAPI/examples  folder ! 
(Most of the example scripts  are blocking  at  : world = World(client.get_world()  )

My question is :  How can I update my scripts to  the latest Carla version ? 

Thank you  for your help  in advance

qhaas, 2019-07-03:
From my team's experience, the PythonAPI of the development(latest) version of CARLA (i.e. 0.9.x series) is in flux, development snapshots (e.g. 0.9.5) do not have detailed/up-to-date documentation in some cases and the PythonAPI changes between snapshots (e.g. from 0.9.2 to 0.9.5).  The CARLA developers warn us of this and it is to be expected for any software under activate development outside a stable branch.  Folks on my team examine the source code of the PythonAPI function calls to determine how to adapt their code to updates between snapshots when the documentation isn't available, which can be tedious, but is the 'price' for using non-stable releases.

In general, do not use the 'nightly build', head of trunk, master branch, etc. of most software projects as they are, in general, subject to significant change and aren't meant for production.  Tagged (snapshot) releases usually are much less of a headache.

For greatest API stability and documentation, the 'stable' 0.8.x branch is likely your best best.  The snapshots of the 0.9 development branch (e.g. 0.9.5) provide a nice balance of new features and documentation/stability, especially when compared to the head of master branch.

remitoudic, 2019-07-05:
Thank you  very much for sharing your experience.  As a young software developer , I appreciate a lot. 
I will follow your advice . I stay with the  0.9.5  , a  nice balance of new features and documentation/stability ... for the rest  there are always a workarounds .

closed 2019-07-05 by remitoudic