carla-simulator/carla #1840: 0 FPS for agent trained on DQN after hundreds of Episodes

Derekabc, 2019-07-03:
Nowadays, I train the ego car based on DQN model. At the beging, the FPS is 20fps for both server and client, and the agent is trying to learn something as the score is better and better. While after hundreds of episode, the FPS of the cilent drops to 0 and the vehicle just stops at the start point. I do not know what is the reason for this.
I start CarlaUE use the command. 
`./CarlaUE4.sh /Game/Carla/Maps/JunctionnoFencev2  -benchmark -fps=20 -windowed -ResX=400 -ResY=300 -carla-port=8000 -carla-server`
At the begaining, the ego car runs smoothly and the agent learns something.
[image]
However, after hundreds of episode, the FPS of the client drops to 0. the vehicle just stops at the start point. 
[image]
We can see the memory and GPU utilization are fine. I do not know what is the reason for this problem. Hope someone can help me or give me some hints for what may cause this problem? Network Topology?

kraken24, 2019-07-08:
@Derekabc can you explain what is your use case for DQN agent? I use DQN agent for developing driving assistance system.

kraken24, 2019-07-08:
Hey I also got same error:
[image]

Derekabc, 2019-07-08:
@kraken24  I designed the DQN for lane follow and lane change tasks.  I do not know how to solve this problem. Now I save all the data and continue to train after it crashes.

lijecaru, 2019-08-04:
Hi @Derekabc can you maybe provide some information on how to implement DQN in Carla?

Derekabc, 2019-08-05:
You can find some information online for how to create Gym-like Carla environment, Then you can build your own agent and interact with the customized environment.

FelippeRoza, 2019-09-20:
I am not sure of how is your implementation, but the rendering at the client side is affected by the DQN processing if both happen in the "same loop".

FelippeRoza, 2019-09-20:
I am not sure of how is your implementation, but the rendering at the client side is affected by the DQN processing if both happen in the "same loop".  

At [https://github.com/FelippeRoza/carla-rl](https://github.com/FelippeRoza/carla-rl) you can find an implementation where the rendering loop occurs in a separate process from the DQN processing (which I called control loop). Hope it helps somehow.

closed 2020-06-26 by germanros1987