carla-simulator/carla #1843: Link error to UE4Editor-Carla.lib

abol-karimi, 2019-07-03:
I can compile and run everything. However, everything I change my C++ code (which is in my own plugin TrafficMonitor as a git submodule) I have to first close the Unreal Editor and compile inside VS. If I try to compile while Unreal Editor is open, I get the following error:

`Parsing headers for CarlaUE4Editor
  Running UnrealHeaderTool "C:\Files\carla\Unreal\CarlaUE4\CarlaUE4.uproject" "C:\Files\carla\Unreal\CarlaUE4\Intermediate\Build\Win64\CarlaUE4Editor\Development\CarlaUE4Editor.uhtmanifest" -LogCmds="loginit warning, logexit warning, logdatabase error" -Unattended -WarningsAsErrors -installed
Reflection code generated for CarlaUE4Editor in 5.3196383 seconds
Using Visual Studio 2017 14.16.27023 toolchain (C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023) and Windows 10.0.17763.0 SDK (C:\Program Files (x86)\Windows Kits\10).
Building 8 actions with 12 processes...
  [1/8] UE4Editor-CarlaUE4-1195.lib
     Creating library C:\Files\carla\Unreal\CarlaUE4\Intermediate\Build\Win64\UE4Editor\Development\CarlaUE4\UE4Editor-CarlaUE4-1195.lib and object C:\Files\carla\Unreal\CarlaUE4\Intermediate\Build\Win64\UE4Editor\Development\CarlaUE4\UE4Editor-CarlaUE4-1195.exp
  [2/8] UE4Editor-Carla-0002.lib
     Creating library C:\Files\carla\Unreal\CarlaUE4\Plugins\Carla\Intermediate\Build\Win64\UE4Editor\Development\Carla\UE4Editor-Carla-0002.lib and object C:\Files\carla\Unreal\CarlaUE4\Plugins\Carla\Intermediate\Build\Win64\UE4Editor\Development\Carla\UE4Editor-Carla-0002.exp
  [3/8] UE4Editor-CarlaUE4-1195.dll
  LINK : fatal error LNK1181: cannot open input file 'C:\Files\carla\Unreal\CarlaUE4\Plugins\Carla\Intermediate\Build\Win64\UE4Editor\Development\Carla\UE4Editor-Carla.lib'
  [4/8] UE4Editor-Carla-0002.dll
     Creating library C:\Files\carla\Unreal\CarlaUE4\Plugins\Carla\Intermediate\Build\Win64\UE4Editor\Development\Carla\UE4Editor-Carla-0002.suppressed.lib and object C:\Files\carla\Unreal\CarlaUE4\Plugins\Carla\Intermediate\Build\Win64\UE4Editor\Development\Carla\UE4Editor-Carla-0002.suppressed.exp
  [5/8] Module.TrafficMonitor.cpp
  [6/8] UE4Editor-TrafficMonitor-0002.lib
     Creating library C:\Files\carla\Unreal\CarlaUE4\Plugins\TrafficMonitor\Intermediate\Build\Win64\UE4Editor\Development\TrafficMonitor\UE4Editor-TrafficMonitor-0002.lib and object C:\Files\carla\Unreal\CarlaUE4\Plugins\TrafficMonitor\Intermediate\Build\Win64\UE4Editor\Development\TrafficMonitor\UE4Editor-TrafficMonitor-0002.exp
  [7/8] UE4Editor-TrafficMonitor-0002.dll
     Creating library C:\Files\carla\Unreal\CarlaUE4\Plugins\TrafficMonitor\Intermediate\Build\Win64\UE4Editor\Development\TrafficMonitor\UE4Editor-TrafficMonitor-0002.suppressed.lib and object C:\Files\carla\Unreal\CarlaUE4\Plugins\TrafficMonitor\Intermediate\Build\Win64\UE4Editor\Development\TrafficMonitor\UE4Editor-TrafficMonitor-0002.suppressed.exp
`
I'm on Windows 10, UE4.22.3, and on CARLA's master branch.

shinhwap, 2019-08-13:
+1
I have the same problem.
Still not solution.

shinhwap, 2019-08-20:
@abol-karimi 
I found the solution for plugin hot reload.
See https://reurl.cc/LvlO9
All we have to do is open up Carla.cpp in plugins and change this line:
`IMPLEMENT_MODULE(FCarlaModule, Carla)`
to this:
`#if WITH_EDITOR
IMPLEMENT_PRIMARY_GAME_MODULE(FDefaultGameModuleImpl, Carla, "Carla")
#else
IMPLEMENT_MODULE(FCarlaModule, Carla)
#endif`
It should solve the plugins hot reload errors.

abol-karimi, 2019-08-23:
It worked! Thanks so much.

closed 2019-08-23 by abol-karimi