carla-simulator/carla #1847: MAP creation without using RoadRunner

arvindpalkyutech, 2019-07-04:
Can I make a Map in CARLA without RoadRunner just using opendrive file. Actually I want to make same Map in CARLA as in OpenStreetMap. I can export OpenStreetMap to Opendrive format using SUMO simulator. Can Anyone guide me how to make fbx file and use it to create MAP in CARLA.

xiaoyang666, 2019-07-04:
I also have the same problem

PatrickFSLin, 2019-07-05:
I use CityEngine to generate OpenStreetMap (.osm file) + fbx file
And convert osm file to OpenDrive (.xodr) by SUMO
However, in Carla MapBuilder is WIP, so importing the files currently cannot work...

arvindpalkyutech, 2019-07-05:
> I use CityEngine to generate OpenStreetMap (.osm file) + fbx file
> And convert osm file to OpenDrive (.xodr) by SUMO
> However, in Carla MapBuilder is WIP, so importing the files currently cannot work...

Thank You Patrik for this very useful information.

xiaoyang666, 2019-07-06:
> > I use CityEngine to generate OpenStreetMap (.osm file) + fbx file
> > And convert osm file to OpenDrive (.xodr) by SUMO
> > However, in Carla MapBuilder is WIP, so importing the files currently cannot work...
> 
> Thank You Patrik for this very useful information.

I have had the same problem before, but I have not tried it yet.This is a reply from a former classmate.

OpenDrive is not a commercial format, but software to generate .xodr seems to be commercial in many cases. This link may be the best open source I've found : https://gitlab.com/OpenRoadEd/OpenRoadEd.

arvindpalkyutech, 2019-07-08:
> I have had the same problem before, but I have not tried it yet.This is a reply from a former classmate.
>
> OpenDrive is not a commercial format, but software to generate .xodr seems to be commercial [...]

Thank You xiaoyang thesis was worth reading.

0102TH, 2019-07-14:
@PatrickFSLin 
That's a good idea.
But the algorithm of sumo and carla is different.
Maybe when the file type is converted into *.xodr by sumo,the *.xodr ‘s geometry is paramPoly3，however，the geometry of carla’s .xodr is Arc and Line.
In my opinion,we should realize the format conversion between paramPoly3 and Arc，Line.
[image]

PatrickFSLin, 2019-07-15:
> @PatrickFSLin
> That's a good idea.
> But the algorithm of sumo and carla is different.
> Maybe when the file type is converted into *.xodr by sumo,the *.xodr ‘s geometry is paramPoly3，however，the [...]

@0102TH That's a good suggestion. Either we realize the format conversion between ParamPoly3 and Arc/Line or ParamPoly3 is directly supported in Carla (that is also my reason of waiting for the new update of Carla).

xiaoyang666, 2019-07-15:
@PatrickFSLin
Thank you for your reply, this is very useful. There are no easy-to-use open source tools for converting paramPoly3 and Arc/Line. I want to create a map on the highway. At present, Roadrunner seems to be unable to try the software. I can only change it on the original map of carla.

xiaoyang666, 2019-07-15:
@PatrickFSLin
Thank you for your reply, this is very useful. Are there any easy-to-use open source tools for conversion of paramPoly3 and Arc/Line? I want to create a map on the highway. At present, Roadrunner seems to be unable to try the software. I can only change it on the original map of carla.

remitoudic, 2019-07-30:
Hello guys , 

Thank you for sharing your experiences .

I would like  to accomplish the same :   convert  an  OSM map   to  xodr / Fbx files  for  Carla .  
Is there any update since 2 weeks ? 

@PatrickFSLin  and  @0102TH did you manage to do the conversion ?  

Thank you in advance for sharing .  :)

closed 2019-10-10 by stale[bot]

cdefg, 2022-09-06:
> @0102TH That's a good suggestion. Either we realize the format conversion between ParamPoly3 and Arc/Line or ParamPoly3 is directly supported in Carla (that is also my reason of waiting for the new [...]

Now It support the conversion, I'll test it for I have the same problem.

saicsk-deloitte, 2023-02-14:
> I use CityEngine to generate OpenStreetMap (.osm file) + fbx file And convert osm file to OpenDrive (.xodr) by SUMO However, in Carla MapBuilder is WIP, so importing the files currently cannot work...

Hi @PatrickFSLin ,

-  What are supporting exporting extensions using City Engine?
- Do you think - It's good to use City Engine for Carla Maps (OSM , FBX formats )?
- Have you found any other tools (except Road Runner & City Engine) for making Carla maps?

evoevo07, 2024-03-26:
Are there any news? I am also trying creating maps without Roadrunner...