carla-simulator/carla #1848: Geolocation to world coordinates

n1try, 2019-07-04:
While the Python API offers `carla.Map.transform_to_geolocation()`, I couldn't find a method to do a conversion the other way round. How can I achieve that?

nsubiron, 2019-07-05:
Sadly that's not implemented.

n1try, 2019-07-05:
I'd appreciate if someone could implement that! I guess is not too big of a change and would help me a lot!

germanros1987, 2020-06-26:
@Axel1092 0.9.11 task?

jht1759, 2020-08-19:
Hello Gentle:
@germanros1987 

I'd like to know. before the official version 0.9.11 release for this issue. how can we do some script for it by ourself. and then we can solve the issue temporary.

Thanks a lot!

jht1759, 2020-08-19:
@muety : Do you have any idea to solve the issue you request? Thanks a lot!

muety, 2020-08-19:
@jht1759 There has to be some kind of linear mapping from world coordinates to GNSS coordinates, i.e. vectors `A` and `B`, such that `P_world = P_gnss * A + B`. However, I don't know how to find those factors in Carla's code base. To be honest, to approximately solve my problem, I sampled a few thousand `(P_world, P_gnss)` pairs and then did a regression to find out their relationship (i.e `A` and `B`), but that's not recommendable as it's not exact and a very ugly hack.

Axel1092, 2020-08-26:
Hi all, we will implement this feature for 0.9.11. In the mean time, you could search on how to perform the projections from geolocation coordinates to the plane X,Y for instance using the mercator projection. There is also the proj library (https://proj.org/) which has a number of projections from geolocation.

rhklite, 2020-12-11:
I'm having trouble with this aswell.  I tried using the python `utm` package: `utm.from_latlon()`, but the result does not match up with the map imported from OSM. Can anyone provide some information on how the (0,0) position in carla is selected when the `.xodr` file is imported?

Sultan91, 2023-09-21:
@Axel1092 Hi, was this feature already implemented?

minhaj6, 2024-07-26:
any update on this issue?

Blyron, 2025-08-07:
This is gonna be achieved for 0.9.16