carla-simulator/carla #1855: vehicle orientation angle

kimiya66, 2019-07-07:
Hi,
How can I get the orientation of the ego vehicle? 
As I saw in one issue you (@nsubiron  @marcgpuig) saide that is yaw, pitch and roll. but when I want to get one value for orientation angle of the car , which attribute or function I can use?
Thanks

Derekabc, 2019-07-10:
you can compute the angle of speed and waypoint direction. See code [https://github.com/carla-simulator/carla/blob/master/PythonAPI/carla/agents/tools/misc.py](url) to get some ideas.

linxigjs, 2019-07-12:
in python, it's world.player.get_transform().rotation.yaw, degree unit. You can find that if read carla/PythonAPI/examples/manual_control.py carefully. The Transform and Rotation classes defination are in carla/LibCarla/source/carla/geom folder.

closed 2019-09-17 by stale[bot]